Xinyuan-LilyGO/LilyGo-Modem-Series #192: Unable to get data from CANBus

jichef, 2024-12-12:
Hello! I am trying to obtain data from a Canbus using an SN65HVD230 Transciever. It works with any ESP32 of 30 or 38 pins, but I can't use the same code in my lilygo-t-A7670G, nor in the TSIM7000G. I have checked the chip review (because if it is 3.x the speed of the canbus must be multiplied by 2).

This is my code:
```
#include <Arduino.h>
#include <SP32Can.H>
#include <can_config.h>
#include "Bluetoothserial.h"

Bluetoothserial serialbt;

Can_device_T Can_cfg;
const int rx_queue_size = 10;

UNSIGNED Long Latestime = 0;
Const Unsigned Long Interval = 20000;

void setup (void) {
 Serial.begin (115200);
 Serialbt.begin ("A7670");

 // Can Bus Configuration
 Can_cfg.speed = can_speed_500kbps;
 Can_cfg.tx_pin_id = gpio_num_32;
 Can_cfg.rx_pin_id = gpio_num_33;
 Can_cfg.rx_queue = xqueuecreate (rx_queue_size, sizeof (can_frame_t));

 // Set Can Filter
 Can_filter_t p_filter;
 p_filter.fm = single_mode;

 p_filter.acr0 = 0x41;
 p_filter.acr1 = 0x00;
 p_filter.acr2 = 0x00;
 p_filter.acr3 = 0x00;

 p_filter.amr0 = 0xff;
 p_filter.amr1 = 0xff;
 p_filter.amr2 = 0xff;
 p_filter.amr3 = 0xff;

 Esp32can.Conconfigfilter (& p_filter);
 Esp32can.caninit ();
 Serial.println ("Can Bus Initialized With Filter for ID 0x41.");
}

void loop (void) {
 Can_frame_t rx_frame;

 if (millis () - ballastime> = interval) {
 Lateadtime = Millis ();

 if (xqueuereiveive (can_cfg.rx_queue, & rx_frame, 3 * porttick_period_ms) == pdtrue) {
 // Send raw can data via serialbt
 Serialbt.print ("id: 0x");
 Serialbt.print (rx_frame.msgid, hex);
 Serialbt.print ("dlc:");
 Serialbt.print (rx_frame.fir.b.dlc);
 Serialbt.print ("data:");

 for (int i = 0; i <rx_frame.fir.b.dlc; i ++) {
 Serialbt.print ("0x");
 Serialbt.print (rx_frame.data.u8 [i], hex);
 if (i <rx_frame.fir.b.dlc - 1) {
 Serialbt.print (",");
 }
 }
 Serialbt.println ();

 // Store the first byte in decimal
 int firstbytedecimal = rx_frame.data.u8 [0];
 Serial.print ("first byte in decimal:");
 Serial.println (FirstbytEdecimal);
 }
 }
}
```
Hello! I am trying to obtain data from a Canbus using an SN65HVD230 Transciever. It works with any ESP32 of 30 or 38 pins, but I can't use the same code in my lilygo-t-A7670G, nor in the TSIM7000G. I have checked the chip review (because if it is 3.x the speed of the canbus must be multiplied by 2).

This is my code:
#include <Arduino.h>
#include <SP32Can.H>
#include <can_config.h>
#include "Bluetoothserial.h"

Bluetoothserial serialbt;

Can_device_T Can_cfg;
const int rx_queue_size = 10;

UNSIGNED Long Latestime = 0;
Const Unsigned Long Interval = 20000;

void setup (void) {
 Serial.begin (115200);
 Serialbt.begin ("A7670");

 // Can Bus Configuration
 Can_cfg.speed = can_speed_500kbps;
 Can_cfg.tx_pin_id = gpio_num_32;
 Can_cfg.rx_pin_id = gpio_num_33;
 Can_cfg.rx_queue = xqueuecreate (rx_queue_size, sizeof (can_frame_t));

 // Set Can Filter
 Can_filter_t p_filter;
 p_filter.fm = single_mode;

 p_filter.acr0 = 0x41;
 p_filter.acr1 = 0x00;
 p_filter.acr2 = 0x00;
 p_filter.acr3 = 0x00;

 p_filter.amr0 = 0xff;
 p_filter.amr1 = 0xff;
 p_filter.amr2 = 0xff;
 p_filter.amr3 = 0xff;

 Esp32can.Conconfigfilter (& p_filter);
 Esp32can.caninit ();
 Serial.println ("Can Bus Initialized With Filter for ID 0x41.");
}

void loop (void) {
 Can_frame_t rx_frame;

 if (millis () - ballastime> = interval) {
 Lateadtime = Millis ();

 if (xqueuereiveive (can_cfg.rx_queue, & rx_frame, 3 * porttick_period_ms) == pdtrue) {
 // Send raw can data via serialbt
 Serialbt.print ("id: 0x");
 Serialbt.print (rx_frame.msgid, hex);
 Serialbt.print ("dlc:");
 Serialbt.print (rx_frame.fir.b.dlc);
 Serialbt.print ("data:");

 for (int i = 0; i <rx_frame.fir.b.dlc; i ++) {
 Serialbt.print ("0x");
 Serialbt.print (rx_frame.data.u8 [i], hex);
 if (i <rx_frame.fir.b.dlc - 1) {
 Serialbt.print (",");
 }
 }
 Serialbt.println ();

 // Store the first byte in decimal
 int firstbytedecimal = rx_frame.data.u8 [0];
 Serial.print ("first byte in decimal:");
 Serial.println (FirstbytEdecimal);
 }
 }
}
```

Using this code in any ESP32 of 30 or 38 pins it works, but not on the Lilygo tsim A7670 or TSIM7000. Any clue?
Thanks!

lewisxhe, 2024-12-13:
1. esp32 does not have GPIO30, only GPIO33~GPIO39, and GPIO33 and above can only be configured as input
2. SIM7000G and A7670 both use esp32 modules, which is no different from other boards. I mean modules
3. You can refer to the CAN example [here](https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/example/Arduino/CAN/CAN.ino), or the official example of [ESP TWAI](https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/TWAI)

jichef, 2024-12-13:
Sorry, what I meant to say with 30 or 38 pins was reafering to the esp model [image]

Anyway, thanks for yogur reply. Ill give a try with official TWAI. 

I'll keep you informed.

closed 2025-01-28 by github-actions[bot]