Xinyuan-LilyGO/LilyGo-Modem-Series #195: AT+CGSN returns memory failure

ArthurTimmermans, 2024-12-15:
Hello,

I wanted to report a bug on the A7670G latest firmware.
If you want to check the IMEI of the device, it returns a memory failure CME

`AT+CGSN   
+CME ERROR: memory failure`

Also the ATI does not return an IMEI

`AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 
`

[image]

lewisxhe, 2024-12-16:
The IMEI may be lost when upgrading the firmware. You can use the

```bash
AT+SIMEI=867284064546817
```
command to restore it

lewisxhe, 2024-12-16:
By the way, thanks for your reply to other issues.

closed 2024-12-16 by ArthurTimmermans