Xinyuan-LilyGO/LilyGo-Modem-Series #196: Error en POST, código HTTP: 713

theChosen16, 2024-12-15:
Hi you all, 

i updated the firmware of my lilygo A7670G / ESP32-WROVER-E

[image]
On the image is showing the version (like in the video firmware update).

I'm trying to send a .json to a server with HTTPS protocol (with wifi works, but no with SIM).

Before the update, i was facing error 715, now i got the 713 error.

What we can do ?

ArthurTimmermans, 2024-12-15:
@theChosen16 Please share your code, so we can see what AT commands you are sending.

Make sure you use SNI and ignorecertCN (or even better, upload a root CA and use it)

theChosen16, 2024-12-15:
@ArthurTimmermans this is my code:

#define TINY_GSM_RX_BUFFER 1024

#include "utilities.h"
#include <TinyGsmClient.h>
#include <ArduinoJson.h>

// Puerto serie para debug
#define SerialMon Serial
#define SerialAT Serial1

// Configuración de APN
const char APN[]  = "bam.clarochile.cl";
const char USER[] = "clarochile";
const char PASS[] = "clarochile";

// Configuración HTTP/HTTPS
const char server[] = "api.digitaldatafarm.com";
const char resource[] = "https://api.digitaldatafarm.com/api/v4/logger/save_sonda_info";
const char bearerToken[] = "eyJhbGciOiJIUzI1NiJ9.eyJlaWQiOiI4NCIsInJvbGUiOiJjbGllbnQiLCJzdWIiOiJBbGVVU00ifQ.rxWjbaw_JHWQg-K0xTUW0ETno9tUo_BuyaIxtIT-MrA";

// Objeto módem
TinyGsm modem(SerialAT);

void setup() {
  SerialMon.begin(115200);
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);

#ifdef BOARD_POWERON_PIN
  pinMode(BOARD_POWERON_PIN, OUTPUT);
  digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif

  // Resetear módem
  pinMode(MODEM_RESET_PIN, OUTPUT);
  digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL); delay(100);
  digitalWrite(MODEM_RESET_PIN, MODEM_RESET_LEVEL); delay(2600);
  digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);

  // Encender módem
  pinMode(BOARD_PWRKEY_PIN, OUTPUT);
  digitalWrite(BOARD_PWRKEY_PIN, LOW);
  delay(100);
  digitalWrite(BOARD_PWRKEY_PIN, HIGH);
  delay(100);
  digitalWrite(BOARD_PWRKEY_PIN, LOW);

  SerialMon.println("Iniciando módem...");
  int retry = 0;
  while (!modem.testAT(1000)) {
    SerialMon.print(".");
    if (retry++ > 10) {
      // Reintentar encendido
      digitalWrite(BOARD_PWRKEY_PIN, LOW);
      delay(100);
      digitalWrite(BOARD_PWRKEY_PIN, HIGH);
      delay(1000);
      digitalWrite(BOARD_PWRKEY_PIN, LOW);
      retry = 0;
    }
  }
  SerialMon.println("");

  // Revisar SIM
  SimStatus sim = SIM_ERROR;
  while (sim != SIM_READY) {
    sim = modem.getSimStatus();
    if (sim == SIM_LOCKED) {
      SerialMon.println("La SIM está bloqueada.");
      return;
    }
    delay(1000);
  }
  SerialMon.println("SIM lista");

  // Esperar registro en red
  SerialMon.print("Esperando registro en la red...");
  RegStatus status = REG_NO_RESULT;
  while (status == REG_NO_RESULT || status == REG_SEARCHING || status == REG_UNREGISTERED) {
    status = modem.getRegistrationStatus();
    if (status == REG_DENIED) {
      SerialMon.println("Registro denegado. Revisa APN.");
      return;
    }
    delay(1000);
  }
  SerialMon.println("Registrado en red.");

  // Activar red
  if (!modem.setNetworkActive()) {
    SerialMon.println("Error al activar red.");
    return;
  }

  // Conectar GPRS
  if (!modem.gprsConnect(APN, USER, PASS)) {
    SerialMon.println("Error al conectar GPRS.");
    return;
  }
  SerialMon.println("GPRS conectado.");

  String ipAddress = modem.getLocalIP();
  SerialMon.print("IP de red: ");
  SerialMon.println(ipAddress);

  // Configurar SNI y versión TLS antes de iniciar HTTPS
  modem.sendAT("+CSSLCFG=\"enableSNI\",0,1");
  modem.waitResponse();
  modem.sendAT("+CSSLCFG=\"sslversion\",0,4"); // TLS 1.2
  modem.waitResponse();

  // Iniciar HTTPS
  modem.https_begin();
}

void loop() {
  // Crear el JSON
  StaticJsonDocument<512> doc;
  doc["v"] = "0.1";
  doc["sn"] = "A00002120B";
  
  JsonArray dl_entries = doc.createNestedArray("dl_entries");
  JsonObject entry = dl_entries.createNestedObject();
  entry["ts"] = 1731687600; 
  JsonObject aleusm = entry.createNestedObject("ALEUSM");
  JsonObject node = aleusm.createNestedObject("1");
  JsonArray me = node.createNestedArray("ME");
  me.add("00");
  me.add(25.0);  
  me.add(60.0);  
  me.add(0.0);   

  String jsonString;
  serializeJson(doc, jsonString);

  SerialMon.println("JSON generado:");
  SerialMon.println(jsonString);

  // Verificar conexión GPRS
  if (!modem.isGprsConnected()) {
    SerialMon.println("GPRS no conectado, intentando reconectar...");
    if (!modem.gprsConnect(APN, USER, PASS)) {
      SerialMon.println("Error al reconectar GPRS.");
      delay(60000);
      return;
    }
  }

  // Setear la URL HTTPS
  if (!modem.https_set_url(resource)) {
    SerialMon.println("Fallo al establecer URL HTTPS");
    delay(5000);
    return;
  }

  // Agregar headers
  modem.https_add_header("Content-Type", "application/json");
  modem.https_add_header("Authorization", String("Bearer ") + bearerToken);
  modem.https_add_header("Connection", "Close");
  modem.https_add_header("Accept", "*/*");

  // Enviar petición POST
  int httpCode = modem.https_post(jsonString);

  if (httpCode != 200) {
    SerialMon.print("Error en POST, código HTTP: ");
    SerialMon.println(httpCode);
  } else {
    SerialMon.print("HTTP Status: ");
    SerialMon.println(httpCode);

    String header = modem.https_header();
    SerialMon.print("Headers: ");
    SerialMon.println(header);

    String response = modem.https_body();
    SerialMon.print("Respuesta: ");
    SerialMon.println(response);
  }

  delay(60000);
}

ArthurTimmermans, 2024-12-15:
@theChosen16 Add

modem.sendAT("+CSSLCFG="ignorecertCN",0,1"); 
modem.waitResponse();

or add the rootCA to your device

theChosen16, 2024-12-16:
adding this lines i reach the connection via sim with the server, thanks you a lot @ArthurTimmermans 

void sendData(String json) { 
    modem.https_begin(); 
   
    modem.sendAT("AT+CSSLCFG=\"ignorecertCN\",0,1");

    modem.waitResponse();

closed 2024-12-16 by theChosen16