Xinyuan-LilyGO/LilyGo-Modem-Series #197: Error when subscribing to MQTT topic with modem firmware B05

0xMattijs, 2024-12-16:
This is a follow-up on #183 

After upgrading the modem firmware to B05 subscribing to a topic fails.

Here's the AT command log of the `MqttsBuiltinHivemq` example:

```
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="enableSNI",0,1
OK
AT+CSSLCFG="authmode",0,1
OK
AT+CMQTTREL=0
+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"TSIM7670-2",1
OK
AT+CMQTTCFG="version",0,4
OK
AT+CMQTTCONNECT=0,"tcp://[redacted]"
OK

+CMQTTCONNECT: 0,0successfully.

AT+CMQTTDISC?
+CMQTTDISC: 0,0
+CMQTTDISC: 1,1
OK
MQTT has connected!
AT+CMQTTSUBTOPIC=0,21,0
ERROR
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish
GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
>RunTime:14
RunTime:14

OK
AT+CMQTTPUB=0,0,60,0,0
OK

+CMQTTPUB: 0,0
```

lewisxhe, 2024-12-16:
Oh my god, another problem? 😱I will test it tomorrow. You can try downgrading to an older version first.

0xMattijs, 2024-12-16:
@lewisxhe I downgraded to B02 with the same result. I didn't find a download link for B01, so I was unable to test.

lewisxhe, 2024-12-17:
It looks like something has been changed, I have already sent it to SIMCOM.
[image]

0xMattijs, 2024-12-18:
OK. I think I found the culprit. Apparently the `CMQTTSUBTOPIC` was removed and setting the topic is now a part of the `CMQTTSUB` command, similar to the SIM7028 module: https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/unit/Unit%20NB-IoT2%20(SIM7028)/SIM7028%20Series_MQTT(S)_Application%20Note_V1.03.pdf
I patched the TinyGSM code and now it successfully subscribes:

```
AT+CMQTTTOPIC=0,19
>GsmMqttTest/publish
GsmMqttTest/publish
OK
```

lewisxhe, 2024-12-18:
OK. The problem seems to be with me. I wrote it based on A7670 at the beginning. I didn't look carefully when testing SIM7670G. CMQTTSUBTOPIC has been deleted here in SIM7670G. Thank you for your test. I'm sorry for my negligence. Thank you

lewisxhe, 2024-12-18:
For A7670 and A7608, the CMQTTSUBTOPIC command should no longer be used. I just tested A7670 and it is normal.

0xMattijs, 2024-12-18:
OK. In that case my PR will break A7670 and A7608. Seems we need to test for h/w model in the respective code

lewisxhe, 2024-12-18:
No, they are compatible.

closed 2024-12-20 by 0xMattijs