Xinyuan-LilyGO/LilyGo-Modem-Series #199: T-pcie A7670G differents versions one work the other randomly restart

fbp2m, 2024-12-19:
Hi,

For our product we have purchased first time T-PCIE A7670G.
We received A7670G-LABE with firmware A069B01A7670M7 and pcb AX7670 V1.0 21-11-28
These modems work well on our product and software. (We also use SIM7070G, SIM7600E and SIM7600G-H without any problem).

We have placed a new order of A7670G (about 200pcs), but this time we received A7670G-LLSE with firmware A011B20A7670M7 and pcb AX7670 V1.1 23-11-15.
These modems randomly restart during their use.
We also try this modem in T_PCIE_V1.1 with same issue. We suspect power so tries are done with 5V3A power supply but restart continue.
But if it was power supply issue why all others modem works ?
Is this a known issue ?
Is there a firmware update for A7670G-LLSE ?

The problem looks like issue #172 but there no details on modem version.
Best regards

lewisxhe, 2024-12-19:
What is the firmware version that won't reboot? Send AT+SIMCOMATI to check what the two versions are.

fbp2m, 2024-12-19:
AT+SIMCOMATI from version won't reboot : 
```
Manufacturer: INCORPORATED
Model: A7670G-LABE
Revision: A069B01A7670M7
A7670M7_B01V02_221012
QCN:
IMEI: 8672840604XXXXX
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
```

AT+SIMCOMATI from reboot : 
```
Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A011B20A7670M7
A7670M7_B20V01_240325
QCN:
IMEI: 8672840654XXXXX
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
```

For more informations, we use it via UART (no USB) at baudrate 1842000 ( 3686400 does not work ) with hardware flow control.

lewisxhe, 2024-12-19:
Try to update A7670G-LLSE to the latest version for testing.

[FIRMWARE](https://drive.google.com/file/d/1-UOq_7TxU8aNAY4KKuXtnoRt0UZwipSQ/view)

[How upgrade](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE/blob/master/doc/sim7600SA_update_firmware.md)

[Same board](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/178)

lewisxhe, 2024-12-19:
I have a question, the 200 you received this time were both versions available? Or a separate one?

lewisxhe, 2024-12-19:
Can you take a picture of the LABE version of the board for me to confirm.

fbp2m, 2024-12-19:
I made a mistake, I update the wrong module, LABE instead of LLSE...
So the LABE does not work anymore.
After that, I update with new firmware on right LLSE.
But after update I lost IMEI (on both LLSE and LABE) : 
```
Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI:

OK
```
AT+SIMEI? give me an error.
So I put it with AT+SIMEI=867284065474795
Now IMEI get back in AT+SIMCOMATI
```
Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 867284065474795

OK
```
But unfortunately I got reboots again.

We make an order of 200pcs, only 12 were in stocks, we received it which are LABE and work flawless.
After we received 188 others which are LLSE and we get troubles with.

This is the picture of LABE (I broke with wrong update)
Do you have firmware for the LABE so I can try to repair it?

[image]

And the LLSE : 
[image]

lewisxhe, 2024-12-20:
When did you buy it?
Did you buy 200 at once? Do you have both versions?

I checked LABE and it should be a very old version

lewisxhe, 2024-12-20:
Regarding the LLSE restart issue, I need to get the board on Monday to test it. I will check if I have the same situation here.

lewisxhe, 2024-12-20:
Please contact the dealer you purchased from, we need to trace the production batch

lewisxhe, 2024-12-20:
You should have used 4G TO AP, right? This requires a huge current. I will test it on Monday. But I personally think it is caused by unstable power supply.

fbp2m, 2024-12-20:
> You should have used 4G TO AP, right? This requires a huge current. I will test it on Monday. But I personally think it is caused by unstable power supply.

Yes, that's the first thing I thought of. This is why we used a 5V3A power supply. But which raises the question, why we do not have the problem with the other modules that we have: A7670G-LABE, SIM7070G, SIM7600E and SIM7600G-H.

lewisxhe, 2024-12-20:
OK, I will test it on Monday. I may need the modem log to analyze the cause of the reboot. This may be tricky.

fbp2m, 2024-12-20:
Thanks for your help !

Since I made mistake when I update LABE from A069B01A7670M7 to A124B01A7670M7 LLSE and put back to A110B06A7670M7 LABE, I got same reboot issue on LABE.
Did you have original firmware A069B01A7670M7 for LABE (I did not find it), I can try if with this original version fix back the reboot problem. 
This would be a good clue to follow.

lewisxhe, 2024-12-21:
I need to check

lewisxhe, 2024-12-23:
> We bought 200 two month ago, but only 12 were in stock. We received it, they are LABE and we received other 188, 3-4 weeks later which are LLSE.
> 
> Regarding the LLSE restart problem, as it is random to increase the chance of reproducing it, I use example ap2pppos from esp-idf. I just publish my own example on this repo : https://github.com/fbp2m/t-pcie-ap2pppos

I tested this code, I can't run this code.

I tested the current Arduino ESP32 ppp example, tested the LLSE without restarting, you can try

```
#include <PPP.h>
#include <WiFi.h>

#define PPP_MODEM_APN "CHN-CT"
#define PPP_MODEM_PIN "0000"  // or NULL

// WaveShare SIM7600 HW Flow Control
#define PPP_MODEM_RST -1
#define PPP_MODEM_RST_LOW true  //active HIGH
#define PPP_MODEM_RST_DELAY 200
#define PPP_MODEM_TX 27
#define PPP_MODEM_RX 26
#define PPP_MODEM_RTS -1
#define PPP_MODEM_CTS -1
#define PPP_MODEM_FC ESP_MODEM_FLOW_CONTROL_NONE
#define PPP_MODEM_MODEL PPP_MODEM_SIM7600

// WiFi Access Point Config
#define AP_SSID "ESP32-ETH-WIFI-BRIDGE"
#define AP_PASS "password"

IPAddress ap_ip(192, 168, 4, 1);
IPAddress ap_mask(255, 255, 255, 0);
IPAddress ap_leaseStart(192, 168, 4, 2);
IPAddress ap_dns(8, 8, 4, 4);
#define BOARD_PWRKEY_PIN                    (4)
void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  pinMode(25, OUTPUT);
  digitalWrite(25, HIGH);
  delay(100);

  pinMode(BOARD_PWRKEY_PIN, OUTPUT);
  digitalWrite(BOARD_PWRKEY_PIN, LOW);
  delay(100);
  digitalWrite(BOARD_PWRKEY_PIN, HIGH);
  delay(100);
  digitalWrite(BOARD_PWRKEY_PIN, LOW);
  delay(20000);
  // Listen for modem events
  Network.onEvent(onEvent);

  // Start the Access Point
  WiFi.AP.begin();
  WiFi.AP.config(ap_ip, ap_ip, ap_mask, ap_leaseStart, ap_dns);
  WiFi.AP.create(AP_SSID, AP_PASS);
  if (!WiFi.AP.waitStatusBits(ESP_NETIF_STARTED_BIT, 1000)) {
    Serial.println("Failed to start AP!");
    return;
  }

  // Configure the modem
  PPP.setApn(PPP_MODEM_APN);
  PPP.setPin(PPP_MODEM_PIN);
  PPP.setResetPin(PPP_MODEM_RST, PPP_MODEM_RST_LOW, PPP_MODEM_RST_DELAY);
  PPP.setPins(PPP_MODEM_TX, PPP_MODEM_RX, PPP_MODEM_RTS, PPP_MODEM_CTS, PPP_MODEM_FC);

  Serial.println("Starting the modem. It might take a while!");
  PPP.begin(PPP_MODEM_MODEL);

  Serial.print("Manufacturer: ");
  Serial.println(PPP.cmd("AT+CGMI", 10000));
  Serial.print("Model: ");
  Serial.println(PPP.moduleName());
  Serial.print("IMEI: ");
  Serial.println(PPP.IMEI());

  bool attached = PPP.attached();
  if (!attached) {
    int i = 0;
    unsigned int s = millis();
    Serial.print("Waiting to connect to network");
    while (!attached && ((++i) < 600)) {
      Serial.print(".");
      delay(100);
      attached = PPP.attached();
    }
    Serial.print((millis() - s) / 1000.0, 1);
    Serial.println("s");
    attached = PPP.attached();
  }

  Serial.print("Attached: ");
  Serial.println(attached);
  Serial.print("State: ");
  Serial.println(PPP.radioState());
  if (attached) {
    Serial.print("Operator: ");
    Serial.println(PPP.operatorName());
    Serial.print("IMSI: ");
    Serial.println(PPP.IMSI());
    Serial.print("RSSI: ");
    Serial.println(PPP.RSSI());
    int ber = PPP.BER();
    if (ber > 0) {
      Serial.print("BER: ");
      Serial.println(ber);
      Serial.print("NetMode: ");
      Serial.println(PPP.networkMode());
    }

    Serial.println("Switching to data mode...");
    PPP.mode(ESP_MODEM_MODE_CMUX);  // Data and Command mixed mode
    if (!PPP.waitStatusBits(ESP_NETIF_CONNECTED_BIT, 1000)) {
      Serial.println("Failed to connect to internet!");
    } else {
      Serial.println("Connected to internet!");
    }
  } else {
    Serial.println("Failed to connect to network!");
  }
}

void loop() {
  delay(20000);
}

void onEvent(arduino_event_id_t event, arduino_event_info_t info) {
  switch (event) {
    case ARDUINO_EVENT_PPP_START: Serial.println("PPP Started"); break;
    case ARDUINO_EVENT_PPP_CONNECTED: Serial.println("PPP Connected"); break;
    case ARDUINO_EVENT_PPP_GOT_IP:
      Serial.println("PPP Got IP");
      Serial.println(PPP);
      WiFi.AP.enableNAPT(true);
      break;
    case ARDUINO_EVENT_PPP_LOST_IP:
      Serial.println("PPP Lost IP");
      WiFi.AP.enableNAPT(false);
      break;
    case ARDUINO_EVENT_PPP_DISCONNECTED:
      Serial.println("PPP Disconnected");
      WiFi.AP.enableNAPT(false);
      break;
    case ARDUINO_EVENT_PPP_STOP: Serial.println("PPP Stopped"); break;

    case ARDUINO_EVENT_WIFI_AP_START:
      Serial.println("AP Started");
      Serial.println(WiFi.AP);
      break;
    case ARDUINO_EVENT_WIFI_AP_STACONNECTED: Serial.println("AP STA Connected"); break;
    case ARDUINO_EVENT_WIFI_AP_STADISCONNECTED: Serial.println("AP STA Disconnected"); break;
    case ARDUINO_EVENT_WIFI_AP_STAIPASSIGNED:
      Serial.print("AP STA IP Assigned: ");
      Serial.println(IPAddress(info.wifi_ap_staipassigned.ip.addr));
      break;
    case ARDUINO_EVENT_WIFI_AP_PROBEREQRECVED: Serial.println("AP Probe Request Received"); break;
    case ARDUINO_EVENT_WIFI_AP_STOP: Serial.println("AP Stopped"); break;

    default: break;
  }
}

```

fbp2m, 2024-12-23:
Thanks for firmware.
I'm currently on Christmas leave so I can't test it.
In your arduino test did you try to increase modem baudrate to speedup connection ?
Restart occur randomly, to increase chances of this happening I set baudrate to 1842000 (3686400 does not work with my A7670G, only with SIM7600x) and I do somes speedtest from a device connected to AP.
Without this the reboot is quite rare.

I will try firmware and Arduino test next week when I am back.

fbp2m, 2024-12-29:
Hi,

I put the firmware A069B01 back on A7670G-LABE, and after many try I didn't get another reboot.
So I think there is a problem with the new simcom firmware used in our specific case.

I also test your arduino example.
I just add this line at line 69 :
`PPP.setBaudrate(1842000);`

And I get same random restart of modem with LLSE and not with LABE (with firmware A069B01).
There are many errors like this, but it works, before modem restart : 
`E (164114) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1`

lewisxhe, 2024-12-30:
1842000 baud rate? Oh! I suspect the baud rate is too high. Have you tested 115200?

fbp2m, 2024-12-30:
Yes, I tested 115200, but connexion speed is very slow, I can't do real stress test.
1842000 baud rate works with A7670-LABE (with firmware A069B01).
On SIM7600 I use 3686400 baud rate (this does not work on A76070 unlike what is written in the simcom documentation).

In arduino code, I forgot to say that I have changed ESP_MODEM_MODE_CMUX to ESP_MODEM_MODE_DATA  
for the example to work properly.

lewisxhe, 2024-12-30:
I personally feel that the baud rate is too high. In this case, you can only capture the modem log and seek SIMCOM's analysis. You need to provide the log. For specific methods, please see here. https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/a7670x_capture_log.md

fbp2m, 2024-12-30:
Ok I understand that the speed may be too fast.
What I don't understand is that this worked until now with all our other modems, we only have this problem with the latest A7670-LLSEs received.
I managed to capture the logs during a reboot (after 12 mins).
[image]

I put export on repo :
https://github.com/fbp2m/t-pcie-ap2pppos/tree/main/SIMCOM_log

lewisxhe, 2024-12-31:
Are you using A7670G-LLSE A124B01 version? Please send AT+SIMCOMATI to check the version and tell me

lewisxhe, 2024-12-31:
I don't use esphome, you can go to the esphome page and submit a feature request to them

fbp2m, 2024-12-31:
This is AT+SIMCOMATI : 
```
Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 867284065474795

OK
```

Since yesterday I reduce baud rate to 921600 (115200 is to slow for our use case), and I have no restart.

lewisxhe, 2025-01-02:
Yes. I have also asked about the high baud rate, and I am waiting for their reply.

closed 2025-02-16 by github-actions[bot]