Xinyuan-LilyGO/LilyGo-Modem-Series #200: Sending GPS data to Firebase

J7a7m2m8y5, 2024-12-19:
Hi guys, I'm new to programming. I want to send my GPS to firebase database. But I keep getting bad request error. I modified the Https client example code from the repository. I just replaced the URL, the resource path and the port. I would be glad if someone can help me with a code. Thank you

lewisxhe, 2024-12-20:
A7672E ? ?
Which board do you use?

J7a7m2m8y5, 2024-12-20:
> A7672E ? ? Which board do you use?

A7672E FASE

lewisxhe, 2024-12-20:
I haven't seen this version. Is it a LilyGo board?
Send AT+SIMCOMATI to check the version

J7a7m2m8y5, 2024-12-20:
[image]

J7a7m2m8y5, 2024-12-20:
My bad😞 I meant T-A7670

lewisxhe, 2024-12-20:
The example you are using does not support https requests. Please use the https request example test. Please make sure that the request URL is accessible before testing.

J7a7m2m8y5, 2024-12-20:
> The example you are using does not support https requests. Please use the https request example test. Please make sure that the request URL is accessible before testing.

Please can you help me with a sample code of sending and requesting data from firebase?

lewisxhe, 2024-12-20:
What is Firebase? I haven't heard of it. Please send me a link to see if it can help you.

J7a7m2m8y5, 2024-12-20:
> What is Firebase? I haven't heard of it. Please send me a link to see if it can help you.

Google firebase database. There's a Firebase URL and Firebase authentication secret code and also a path. The SSL port for firebase is 443

lewisxhe, 2024-12-20:
Google? Well, Google is blocked in my country by the firewall, I can't connect via 4G, so I can't help you.

J7a7m2m8y5, 2024-12-20:
> Google? Well, Google is blocked in my country by the firewall, I can't connect via 4G, so I can't help you.

Okay, thank you so much. I really appreciate it

J7a7m2m8y5, 2024-12-20:
This code seems to work fine through WiFi Client. But I've been getting errors trying to use TinyGSM Library with the Esp32 Firebase library. `#define TINY_GSM_MODEM_A7670
#include "utilities.h"

#define TINY_GSM_RX_BUFFER 1024  // Set RX buffer to 1Kb
#define SerialAT Serial1

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#include <TinyGsmClient.h>
#include <TinyGPSPlus.h>

#include <Arduino.h>
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

//Provide the token generation process info.
#include "addons/TokenHelper.h"
//Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

// Your GPRS credentials, if any
const char apn[] = "web.gprs.mtnnigeria.net";
const char gprsUser[] = "web";
const char gprsPass[] = "web";

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

// The TinyGPSPlus object
TinyGPSPlus gps;

// Insert your network credentials
#define WIFI_SSID "xxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxx"

// Insert Firebase project API Key
#define API_KEY "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

// Insert RTDB URLefine the RTDB URL */
#define DATABASE_URL "https://xxxxxxxxxxxxxxxxxxxxxxxxxxx.com/"

//Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
int count = 0;
bool signupOK = false;

void displayInfo();

void setup() {
  Serial.begin(115200);
  // Turn on DC boost to power on the modem
  pinMode(BOARD_POWERON_PIN, OUTPUT);
  digitalWrite(BOARD_POWERON_PIN, LOW);

  // Set modem reset pin ,reset modem
[... 111 lines cut ...]
    Firebase.RTDB.setFloat(&fbdo, "HIS001/Speed", gps.speed.kmph());

    // Write an Float number on the database path test/float
    Firebase.RTDB.setString(&fbdo, "HIS001/Date", String(gps.date.day()) + "/" + String(gps.date.month()) + "/" + String(gps.date.year()));
    Firebase.RTDB.setString(&fbdo, "HIS001/Time", String(gps.time.hour()) + ":" + String(gps.time.minute()) + ":" + String(gps.time.second()));
    //smartDelay(1000);
  }
  smartDelay(1000);
}

// This custom version of delay() ensures that the gps object
// is being "fed".
static void smartDelay(unsigned long ms) {
  int ch = 0;
  unsigned long start = millis();
  do {
    while (SerialAT.available()) {
      ch = SerialAT.read();
      // Serial.write(ch);
      gps.encode(ch);
    }
  } while (millis() - start < ms);
}

`

lewisxhe, 2024-12-20:
You should use the secure client instead of the unencrypted client
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SecureClient

J7a7m2m8y5, 2024-12-20:
> You should use the secure client instead of the unencrypted client https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/SecureClient

This works 😁😁😁 Thank you

closed 2025-02-04 by github-actions[bot]