Xinyuan-LilyGO/LilyGo-Modem-Series #202: gps issue

toneshoo, 2024-12-28:
[ ] I have read the troubleshooting section of the README

## What type of issue is this?

[ ] Error or problem compiling the library
[ ] Question or request for help
[ ] Hardware issue
[ ] Network issue
[ 1] GPS issue

### What are you using?

Modem:  i am using a7608-e esp32-s3-wroom

### AT+SIMCOMATI command returns  

String: whenever i try to start the gps i get restartgps! and i see the messega every now and then 

### Scenario, steps to reproduce
 active gps info 

### Expected Results
 like i said gps info 

### Actual Results
 OK
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1

OK
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1 
gps restart! 

### Debug and AT Command Logs

lewisxhe, 2024-12-30:
Are you running the GPS_NMEA_Output example?

```
OK
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1

OK
AT+CGNSSPWR?

+CGNSSPWR: 1,0,1   #  The return should be turned on here and it should not jump to restart GPS.
gps restart!
```

toneshoo, 2025-01-17:
Manufacturer: INCORPORATED
Model: A7608E-H
Revision: A7600M7_V5.0
IMEI: 861513064189053
+GCAP: +CGSM,+FCLASS,+DS
THIS IS  my board 
also yes it just restart somhow
this is from the at commands 
***********************************************************

+CPIN: SIM REMOVED
AT
OK
AT+CGNSSPWR?
+CGNSSPWR: 0,0,1

OK
AT+CGNSSINFO
ERROR

lewisxhe, 2025-01-20:
Please run the GPS_NMEA_Output example to see the final result.

closed 2025-03-06 by github-actions[bot]