Xinyuan-LilyGO/LilyGo-Modem-Series #203: Code upload issues.

DJGLTD, 2024-12-31:
find myself stuck, I am getting a 
"Start Modem"
then I'm getting
"."
"."
"."
"."
"play successfully"
within the Microsoft visual studio code. issue is no matter what i upload i don't seem to get anything different, I have attempt to flash the ESP32 and delete everything on it but i still keep coming back. I am booting up using boot and rst and getting a upload pending but can't progress past this point?

Any suggestions

[ x] I have read the troubleshooting section of the README

## What type of issue is this?

[ ] Error or problem compiling the library
[ x] Question or request for help
[ ] Hardware issue
[ ] Network issue
[ ] GPS issue

### What are you using?

Modem: 

[ ] T-A7670X-ESP32
[x ] T-Call-A7670X-V1-0-ESP32
[ ] T-SIM7672G-ESP32S3
[ ] T-A7608X-ESP32
[ ] T-A7608X-S3-ESP32
[ ] T-PCIE-A7670X-ESP32

Examples: 

### AT+SIMCOMATI command returns  

String: 

### Scenario, steps to reproduce

### Expected Results

### Actual Results

### Debug and AT Command Logs

lewisxhe, 2024-12-31:
Have you tried pressing and holding the ESP32 BOOT button and then plugging in the USB? Please note that there is a SIM BOOT button next to it, so don’t press the wrong one.

DJGLTD, 2024-12-31:
yep boot button and RST button are on the same side, the other button is labelled as 7670_boot pressing and holding that while connecting to usb does nothing, i am attempting to flash the demo AT Debug and for some reason still keep getting back to 
"Play Successful"

Output from terminal is as follows when programming

Processing T-Call-A7670X-V1-0 (board: esp32dev; platform: espressif32@6.4.0; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting TextToSpeech.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- StreamDebugger @ 1.0.1
|-- TinyGSM Fork @ 1.0.0
Building in release mode
Compiling .pio\build\T-Call-A7670X-V1-0\src\TextToSpeech.ino.cpp.o
Retrieving maximum program size .pio\build\T-Call-A7670X-V1-0\firmware.elf
Checking size .pio\build\T-Call-A7670X-V1-0\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.6% (used 21584 bytes from 327680 bytes)
Flash: [==        ]  21.6% (used 283077 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM4
Uploading .pio\build\T-Call-A7670X-V1-0\firmware.bin
esptool.py v4.5.1
Serial port COM4
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00055fff...
Compressed 17536 bytes to 12203...
Writing at 0x00001000... (100 %)
Wrote 17536 bytes (12203 compressed) at 0x00001000 in 0.6 seconds (effective 234.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 481.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 885.7 kbit/s)...
Hash of data verified.
Compressed 283440 bytes to 157500...
Writing at 0x00010000... (10 %)
Writing at 0x0001bfb0... (20 %)
Writing at 0x00024abb... (30 %)
Writing at 0x00029e12... (40 %)
Writing at 0x0002f287... (50 %)
Writing at 0x000348ac... (60 %)
Writing at 0x0003ce72... (70 %)
Writing at 0x00047044... (80 %)
Writing at 0x0004c61b... (90 %)
Writing at 0x00051ba5... (100 %)
Wrote 283440 bytes (157500 compressed) at 0x00010000 in 3.6 seconds (effective 631.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================================================================================================================================== [SUCCESS] Took 11.02 seconds ======================================================================================================================================================================

Environment         Status    Duration
------------------  --------  ------------
T-Call-A7670X-V1-0  SUCCESS   00:00:11.019
======================================================================================================================================================================= 1 succeeded in 00:00:11.019 ======================================================================================================================================================================= *  Terminal will be reused by tasks, press any key to close it. 

**Then when i boot it up i get the following.** 

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
.
.
.
.

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start Sketch
Start modem...
.
.
.
.
.

Play successfully.
Play successfully.
Play successfully.
Play successfully.
Play successfully.
Play successfully.
Play successfully.
Play successfully.
Play successfully.
Play successfully.

and it just Continues on doing that for ever

lewisxhe, 2024-12-31:
OK, it was uploaded by platformio!

That's because you didn't change the running example, the default is examples/TextToSpeech

You need to change the src_dir line to run the example. Please see README.

closed 2025-02-13 by github-actions[bot]