Xinyuan-LilyGO/LilyGo-Modem-Series #204: GPS with A7670E problems to get GPS and getting this error "ERROR .AT+CGNSSPWR=1"

carlestecno, 2024-12-31:
[ ] I have read the troubleshooting section of the README

## What type of issue is this?

[ ] Error or problem compiling the library
[ ] Question or request for help
[ ] Hardware issue
[ ] Network issue
[x ] GPS issue

### What are you using?

Modem: 

[x ] T-A7670X-ESP32
[ ] T-Call-A7670X-V1-0-ESP32
[ ] T-SIM7672G-ESP32S3
[ ] T-A7608X-ESP32
[ ] T-A7608X-S3-ESP32
[ ] T-PCIE-A7670X-ESP32

Examples: 

### AT+SIMCOMATI command returns  

String: 

### Scenario, steps to reproduce

### Expected Results

### Actual Results

### Debug and AT Command Logs

carlestecno, 2024-12-31:
I have this problem with all GPS codes. 
Thanks.

lewisxhe, 2024-12-31:
A7670E-LASE No GPS

carlestecno, 2024-12-31:
Ok, thanks. with fase is working fine. Thanks for fast reply

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 860470067525257
+GCAP: +CGSM,+FCLASS,+DS

OK
Enabling GPS/GNSS/GLONASS
AT+CGNSSPWR=1

OK

closed 2024-12-31 by carlestecno