Xinyuan-LilyGO/LilyGo-Modem-Series #205: T-A7670X PPP_WIFI_BRIDGE

Wikingvv, 2025-01-04:
PL
Posiadam moduł z T-A7670G
Dopiero zaczynam przygodę z ESP32 
Potrzebuje dodać hotspota do oprogramowania ale nie wiem jak

EN (translate)
I have a module with T-A7670G
I'm just starting my adventure with ESP32
I need to add a hotspot to the software but I don't know how

Wikingvv, 2025-01-04:
Dokładnie tak: hotspot to jest programy AP WiFi 
Dziękuję za pomoc. Mam tylko jedno pytanie czy czy będzie działał modem 4G?

> I don't know what is a hotspot but if it is an access point you can use this from random nerd tutorials. You don't need to understand all ( good if yes) but good to start
> 
> [https://randomnerdtutorials.com/esp32-access-point-ap-web-server/#more-67766](url)

carlestecno, 2025-01-05:
El mòdem funciona però encara no sé si  4g , 3g o 2g
Ho he preguntat i espero resposta

Moden works fine but I don't know if 4g or less
I've asked about AT commands to know.

chemmex, 2025-01-05:
@Wikingvv I doubt that is possible with T-A7670X. The 4G hotspot requires that the modem's USB port is connected to the USB OTG port of ESP32-S3 which is not implemented on this board. 
There is Espressif ESP-IDF-based example of how this should work: 

https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module

Wikingvv, 2025-01-05:
> @Wikingvv I doubt that is possible with T-A7670X. The 4G hotspot requires that the modem's USB port is connected to the USB OTG port of ESP32-S3 which is not implemented directly on this board. Or you may try to connect them with external cable.
> 
> There is Espressif ESP-IDF-based example of how this should work:
> 
> https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/host/usb_cdc_4g_module

Pod czas kompilacji dostałem oto taki błęd:
CMake Error at ~/espressif/esp-idf/tools/cmake/build.cmake:611 (message):
  ERROR: Because project depends on iot_usbh_modem (^1.0.0) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  ~/espressif/esp-idf/tools/cmake/project.cmake:735 (idf_build_process)
  CMakeLists.txt:6 (project)

chemmex, 2025-01-05:
Nie wiem. You can try opening the issue in that repository

Wikingvv, 2025-01-06:
> usb_cdc_4g_module relies on the two USBs being connected to each other, which is not available on this board. You can try this example.
> 
> These boards are not designed to be used as hotspots, but rather as data transfers, so this won't work very well.
> 
> [issues199](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/199#issuecomment-2558705773)

Ok, ja tak rozumiem że jest wielki problem z udostępnianiem internetu?
Spróbuje podejść do tego tematu z innej pozycji: Jest potrzeba uruchomienia internet radio (https://github.com/karawin/Ka-Radio32) na tym module ale nie wiem jak to zrobić?

closed 2025-01-08 by Wikingvv