Xinyuan-LilyGO/LilyGo-Modem-Series #207: Reset the modem to factory defaults

carlestecno, 2025-01-06:
[ ] I have read the troubleshooting section of the README

## What type of issue is this?

[ ] Error or problem compiling the library
[x ] Question or request for help
[ ] Hardware issue
[ ] Network issue
[ ] GPS issue

### What are you using?

Modem: 

[x ] T-A7670X-ESP32
[ ] T-Call-A7670X-V1-0-ESP32
[ ] T-SIM7672G-ESP32S3
[ ] T-A7608X-ESP32
[ ] T-A7608X-S3-ESP32
[ ] T-PCIE-A7670X-ESP32

Examples: 
I want to reset default values of modem. Do you know how to do?
I changed values to send data over 4g or lte but now I can't use as accespoint.
I changed parameters to send data over 2g but problem doesn't fix.
Thanks in advance.

carlestecno, 2025-01-06:
Updated firmware and using AT&F0 but doesn't work as access point.
Others a7670 work fine but this one when I change AT+CNMP= 38 to CNMP= 2 it can't fix problem restoring default parameters.

lewisxhe, 2025-01-07:
Did you change the RF parameters? Once the RF parameters are modified, they cannot be changed back. What did you change? If you are not sure, send the following command to query several key parameters

1. AT+CNBP?

2. AT+CNMP?

2. AT+CPSI

Before you know the meaning of the command, you must read the parameters and save them before testing, so that you can restore the parameters after setting errors.

lewisxhe, 2025-01-08:
[image]

There is another key parameter you did not provide me, and I don't know which specific model you are using.

lewisxhe, 2025-01-09:
A7670E-FASE This parameter seems to have been changed
+CNBP: 0X0000000000000180,0X0000000000800D5,0X0000

Resend this parameter, and then test whether the network is normal
AT+CNBP=0X0000000000000000,0X000001E200000095,0X0000

carlestecno, 2025-01-09:
Hi,
when I use this command

`modem.sendAT("+CNBP=0X0000000000000000,0X000001E200000095,0X0000");`

```
TERMINAL RESPONSE: 
OK
AT+CNBP=0X0000000000000000,0X000001E200000095,0X0000

AT+CNBP=0X0000000000000000,0X000001E200000095,0X0000
```

but It doesn't fix and restarting modem doesn't keep changes.

carlestecno, 2025-01-09:
Testing same model of board

```
AT
OK
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
IMEI: 862771072418828

OK

AT+CNBP?
+CNBP: 0X0000000000000180,0X00000000000800D5,0X0000

OK
```

The value is the same and works both for sending data and as an access point.

lewisxhe, 2025-01-09:
What access point?
Are you connected to the Internet normally?

carlestecno, 2025-01-09:
I have no problems connecting to the internet.

I use lilygo to send data over 2g or 4g  and, also, as a hotspot to be connected  using any device with Wi-Fi capabilities without the need to connect to your router,

Sending data over 4g or 2g works fine but as hotspot it doesn't work.

lewisxhe, 2025-01-10:
This was mentioned in another post. The board is not designed to be used as a hotspot, but as an IOT sensor data transmission.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/205#issuecomment-2571839907

closed 2025-02-24 by github-actions[bot]