Xinyuan-LilyGO/LilyGo-Modem-Series #210: Deep sleep wake up don't work from ULP

adlerre, 2025-01-16:
Hi Lewis,

First of all, thank you for your work and the many examples. It all helps a lot.

I'm currently having problems waking the device from deep sleep. I've read the documentation and tried many workarounds to get it to work. But without success!

This is my code:

```c

void ulpInit(unsigned int threshold, int wakeupPeriod) {
#if defined(BOARD_BAT_ADC_PIN) && defined(LILYGO_T_A7670) || defined(LILYGO_T_A7608X)
    unsigned int idx = adcPeriphNum(static_cast<gpio_num_t>(BOARD_BAT_ADC_PIN));

    if (idx == 0) {
        unsigned int channel = adcChannelNum(static_cast<gpio_num_t>(BOARD_BAT_ADC_PIN));

        // ULP program
        // @see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ulp_macros.html
        const ulp_insn_t program[] = {
            I_MOVI(R0, 0),
            I_MOVI(R1, 0),

            M_LABEL(1), // do {
            /* */I_MOVI(R0, 200), // R0 = n * 1000 / 5, where n is the number of seconds to delay, 200 = 1 s
            /* */M_LABEL(2), // do {
            /* */ // since ULP runs at 8 MHz
            /* */ // 40000 cycles correspond to 5 ms (max possible delay is 65535 cycles or 8.19 ms)
            /*  */I_DELAY(40000),
            /*  */I_SUBI(R0, R0, 1), // R0 --;
            M_BGE(2, 1), // } while (R0 >= 1); ... jump to label 2 if R0 > 0

            I_ADC(R0, idx, channel),
            M_BGE(3, threshold), // R0 >= threshold (mV) ... jump to label 3
            M_BL(3, 1), // R0 < 1 ... also jump to 3
            M_BX(1), // else start delay again

            M_LABEL(3), // LABEL 3
            /* */I_MOVI(R0, 0),
            /* */ // check is allowed to wake up
            /* */I_RD_REG(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_RDY_FOR_WAKEUP_S, RTC_CNTL_RDY_FOR_WAKEUP_S),
            M_BGE(4, 1), // if allowed to wakeup jump to label 4
            M_BX(3), // else jump to label 3

            M_LABEL(4), // LABEL 4
            /* */I_WAKE(), // WAKEUP ESP32

            I_END(),
            I_HALT() // STOP ULP PROCESS
        };

        size_t ulpSize = sizeof(program) / sizeof(ulp_insn_t);

        // Clear rtc memory
        memset(RTC_SLOW_MEM, 0, CONFIG_ULP_COPROC_RESERVE_MEM);

        adc1_config_channel_atten(static_cast<adc1_channel_t>(channel), ADC_ATTEN_DB_12);
        adc1_config_width(ADC_WIDTH_BIT_12);
        adc1_ulp_enable();

        // Load ULP program
        ESP_ERROR_CHECK(ulp_process_macros_and_load(ULP_START_OFFSET, program, &ulpSize));
        if (wakeupPeriod > 0) {
            ESP_ERROR_CHECK(ulp_set_wakeup_period(0, wakeupPeriod * 1000 * 1000));
        }

#if CONFIG_IDF_TARGET_ESP32
        /* Disconnect GPIO12 and GPIO15 to remove current drain through
         * pullup/pulldown resistors on modules which have these (e.g. ESP32-WROVER)
         * GPIO12 may be pulled high to select flash voltage.
         */
        rtc_gpio_isolate(GPIO_NUM_12);
        rtc_gpio_isolate(GPIO_NUM_15);
#endif // CONFIG_IDF_TARGET_ESP32

        // Run ULP program
        ESP_ERROR_CHECK(ulp_run(ULP_START_OFFSET));

        ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
    }
#endif
}

void deepSleep(uint32_t ms) {
    esp_sleep_enable_timer_wakeup(ms * 1000);
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

    ulpInit();

    esp_deep_sleep_start();
}
```

The wake up works without reading from ADC!

Have you any hints to get this working? Or a solution to wake up from deep sleep on power supply?

Kind Regards
René

lewisxhe, 2025-01-17:
Have you seen my example?
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/ULP_Monitor/ULP_Monitor.ino

adlerre, 2025-01-17:
Yes, this works!

But if you use...
```c
analogReadMilliVolts(BOARD_BAT_ADC_PIN);
```
...in this code, deep sleep wake up isn't working anymore!

adlerre, 2025-01-17:
I got it working...

```diff
diff --git a/examples/ULP_Monitor/ULP_Monitor.ino b/examples/ULP_Monitor/ULP_Monitor.ino
--- a/examples/ULP_Monitor/ULP_Monitor.ino	(revision e49d1fd06132a9215435f9e929b930dba5f1f371)
+++ b/examples/ULP_Monitor/ULP_Monitor.ino	(date 1737102286741)
@@ -104,6 +104,8 @@
 
 void setup()
 {
+	// wait a bit to get console working
+    delay(2000);
     Serial.begin(115200);
 
     uint32_t adc_result = 0;
@@ -146,6 +148,16 @@
     // Run ULP program
     ESP_ERROR_CHECK(ulp_run(ULP_START_OFFSET));
 
+    int i=0;
+    while(i < 3) {
+        uint32_t battery_voltage = analogReadMilliVolts(BOARD_BAT_ADC_PIN);
+    //    uint32_t battery_voltage = adc1_get_raw(static_cast<adc1_channel_t>(ADC_PAD_INDEX));
+        battery_voltage *= 2;
+        Serial.printf("Battery:%umV\n", battery_voltage);
+        ++i;
+        delay(500);
+    }
+
     Serial.println("Sleeping...");
 
     delay(1000);
@@ -155,6 +167,13 @@
     // Set power domain keep open
     esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
 
+    // configure channel for ulp usage
+    adc1_config_channel_atten(static_cast<adc1_channel_t>(ADC_PAD_INDEX), ADC_ATTEN_DB_11);
+    adc1_config_width(ADC_WIDTH_BIT_12);
+    // @FIXME read adc value before ULP enable, fixes reading issues in deep sleep
+    adc1_get_raw(static_cast<adc1_channel_t>(ADC_PAD_INDEX));
+    adc1_ulp_enable();
+
 #if SOC_PM_SUPPORT_RTC_SLOW_MEM_PD
     esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_ON);
 #endif

```
...but not in my project. :-(
I'm still looking.

closed 2025-01-21 by adlerre