Xinyuan-LilyGO/LilyGo-Modem-Series #212: MqttClient.ino Wifi example doesn't work

majuss, 2025-01-20:
[x] I have read the troubleshooting section of the README

## What type of issue is this?

[x] Error or problem compiling the library
[ ] Question or request for help
[ ] Hardware issue
[ ] Network issue
[ ] GPS issue

### What are you using?

Modem: 

[ ] T-A7670X-ESP32
[ ] T-Call-A7670X-V1-0-ESP32
[ ] T-SIM7672G-ESP32S3
[ ] T-A7608X-ESP32
[x] T-A7608X-S3-ESP32
[ ] T-PCIE-A7670X-ESP32

Examples: 

MqttClient.ino

### AT+SIMCOMATI command returns  

String: 

### Scenario, steps to reproduce

//#define TINY_GSM_USE_GPRS false
#define TINY_GSM_USE_WIFI true

### Expected Results

### Actual Results

### Debug and AT Command Logs

majuss, 2025-01-24:
I tried the normal unaltered HttpClient example. And only get this:
```
346312] ### Modem: A7608SA-H
[346312] Modem Name: A7608SA-H
[346329] Modem Info: Manufacturer: INCORPORATED Model: A7608SA-H Revision: A7600M7_V5.0 IMEI: 862733060741991 +GCAP: +CGSM,+FCLASS,+DS
Waiting for network... success
Network connected
Connecting to YourAPN success
GPRS connected
Performing HTTP GET request... failed to connect
```

majuss, 2025-01-24:
I tried everything, also different Sims and provider. I had no success at all. It seems like there is just no signal for GPRS? Is there a way to use LTE?

lewisxhe, 2025-02-06:
Please confirm your APN. If you configure the wrong APN, you will not be able to connect to the network. At least, it seems that European operators currently need to configure APN to access the network. Then run this example. This example will try to connect to the network.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/Network/Network.ino

closed 2025-03-24 by github-actions[bot]