Xinyuan-LilyGO/LilyGo-Modem-Series #213: Firmware A7670E

JohanBeone, 2025-01-21:
Good day, I need to update the firmware on 2 different A7670E boards as I am struggling with the HTTPS issue, discussed in: https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117

 I am not having any success with Simcom. They are not replying to my mail request for firmware and upgrade tools. (I have opened an account with Simcom to obtain this before posting here or contacting them.)
@lewisxhe If possible can you please assist?

Unit 1 QR:
P/N:S2-109ZT-Z32GV;
SN:MP06242470A80DB;
IMEI:862771076133704;
SW:A011B20A7670M7

Unit 2 QR:
P/N:S2-10AAW-Z319S;
SN:MP06241379D1CD3;
IMEI:862771071980281;
BTMAC:5C46B0BA3DF6;
SW:A011B04A7670M7_F

The unit supports USB and FOTA firmware upgrades, I would prefer to do USB firmware upgrades. I assume I would need to use a flashing tool to do so.

Regards

lewisxhe, 2025-01-22:
You didn't fill in the format required by the issue, and you didn't provide me with the specific model. You only told me A7670E. A7670E also has many versions. Send AT+SIMCOMATI to query.

Also, have you read this? https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

JohanBeone, 2025-01-22:
Hi @lewisxhe Thank you for your response, I did not see the above link. I have looked at it now. 

For unit 1 I get:
(Unit 1 does not have GPS)
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670E-LASE
Revision: A011B20A7670M7
A7670M7_B20V01_240325
QCN:
IMEI: 862771076133704
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK

Would This be the same as A7670E-LASE No GPS in the provided link?

For Unit 2 I Get:

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 862771071977204
MEID:
GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK

Would This unit be the same as A7670E-FASE GPS in the provided link?

Thank you once again for your response, I really really appreciate it.

lewisxhe, 2025-01-22:
LASE and FASE are different. You cannot flash the same firmware. You have to flash the firmware of the corresponding model separately.

JohanBeone, 2025-01-22:
Thank you very much for the information and the link. I was able to update the firmware successfully.

closed 2025-01-22 by JohanBeone