Xinyuan-LilyGO/LilyGo-Modem-Series #215: Why is the battery voltage boosted?

majuss, 2025-01-23:
Hello :)

Why are you boosting the battery voltage? The battery protection will cap out at 3.0 V and that is enough to run the ESP32S3. I need a very low power version and my guess is that the battery booster will drain a significant amount of current.

[image]

lewisxhe, 2025-01-24:
This voltage is also supplied to the MODEM.

majuss, 2025-01-24:
Ahh I thought it will also run with 3 V. Thank you for pointing that out.

closed 2025-01-24 by majuss

majuss, 2025-01-24:
But wouldn't it be better to cap the voltage of the booster to 3.8? Is the booster chip capable of handling bypass? The 3.3 V LDO should also be connected to the battery for better power savings, since the lower voltage will increase the LDOs efficiency?