Xinyuan-LilyGO/LilyGo-Modem-Series #216: Problema scheda T-SIM 7672G V1.0

PIERLUIGI1983, 2025-01-27:
I HAVE PURCHASED SEVERAL CARDS, WHEN TRYING THE EXAMPLES PROVIDED, THE SENDING OF SMS WORKS CORRECTLY, WHILE THE EXAMPLE FOR RECEIVING SMS DOESN'T WORK, JUST AS THE EXAMPLE FOR MAKING A NUMBER RING DOESN'T WORK, WHAT CAN I DO?

lewisxhe, 2025-01-27:
Have you read the README ?

[image]

[image]

PIERLUIGI1983, 2025-01-27:
MI STA DICENDO CHE LA SCHEDA NON SUPPORTA I SMS E LE CHIAMATE; SE è COSI PERCHE RIESCO AD IVIARE SMS ?

PIERLUIGI1983, 2025-01-31:
con questo codice, riesci a leggere i SMS in arrivo 
 if (gsmSerial.available()) {
    String response = gsmSerial.readString();
    Serial.println("Risposta dal modulo GSM:");
    Serial.println(response);

    // Controlla se è arrivato un nuovo SMS
    if (response.indexOf("+CMTI:") != -1) {
      int indexStart = response.indexOf(",") + 1;
      int msgIndex = response.substring(indexStart).toInt();

      // Legge il messaggio ricevuto
      readSMS(msgIndex);
    }
  }
}

void readSMS(int msgIndex) {
  Serial.print("Leggo l'SMS all'indice ");
  Serial.println(msgIndex);

  // Invia il comando AT+CMGR per leggere l'SMS
  gsmSerial.print("AT+CMGR=");
  gsmSerial.println(msgIndex);
  delay(1000);

  // Stampa la risposta del modulo GSM
  while (gsmSerial.available()) {
      msg = gsmSerial.readString();
    Serial.println("Contenuto SMS:");
    Serial.println("msg");Serial.println(msg);
  }

}

void sendATCommand(String command) {
  gsmSerial.println(command);
  delay(1000);

  // Stampa la risposta dal modulo GSM
  while (gsmSerial.available()) {
    String response = gsmSerial.readString();
     Serial.println(response);
  }
}

lewisxhe, 2025-02-06:
Only base stations that support SGS servic can use the SMS function. You said you can send SMS, so the base station you use should support SGS servic. Try to write ATDebug for fault debugging. After starting up, send AT OK, wait for SMS Done, then send SMS from another number to the modem to see if there is any message.

closed 2025-03-24 by github-actions[bot]