Xinyuan-LilyGO/LilyGo-Modem-Series #217: A7608X No GPS location

blackrocket, 2025-01-29:
V1.2 WROOM-E A7608SA-H: GNSS seems to power on but cannot get a location. GSM connects OK. Tried re-copying lib directory to Libraries, updated ESP32 boards, ensured digitalWrite(BOARD_POWERON_PIN, HIGH) set, re-flashed sketches, re-installed USB driver, rebooted IDE, rebooted computer, different antennae and locations through window and outside, on battery, on USB power but no location data. Only thing I can't try is voltage on GPS uFL socket, no multimeter available here. Any ideas? Thanks!
********************************************
GPS_NMEA_Output sketch shows:
Start modem...
Modem starting...AT
AT
AT
AT
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: INCORPORATED
Model: A7608SA-H
Revision: A7600M7_V5.0
IMEI: --REMOVED--
+GCAP: +CGSM,+FCLASS,+DS

OK
Model Name:A7608SA-H
Enabling GPS/GNSS/GLONASS
AT+CVAUXS=1

OK
AT+CGNSSPWR=1

OK

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
AT+CGNSSMODE=3
(Stopped here)
*******************************************************
GPS_NMEA_Parse shows:
Start modem...
Modem starting...AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: INCORPORATED
Model: A7608SA-H
Revision: A7600M7_V5.0
IMEI: --REMOVED--
+GCAP: +CGSM,+FCLASS,+DS

OK
Model Name:A7608SA-H
Enabling GPS/GNSS/GLONASS
AT+CVAUXS=1

OK
AT+CGNSSPWR=1

OK

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
AT+CGNSSMODE=3

OK
AT+CGNSSNMEA=1,1,1,1,1,1,0,0

OK
AT+CGPSNMEARATE=1

OK
AT+CGNSSTST=1

OK
AT+CGNSSPORTSWITCH=0,1

OK
Sats HDOP  Latitude   Longitude   Fix  Date       Time     Date Alt    Course Speed Card  Distance Course Card  Chars Sentences Checksum
           (deg)      (deg)       Age                      Age  (m)    --- from GPS ----  ---- to London  ----  RX    RX        Fail
----------------------------------------------------------------------------------------------------------------------------------------
AT+CGNSSPWR?
+CGNSSPWR: 1,**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   9     0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   9     0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   25    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   25    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   37    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   37    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   37    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   37    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   37    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   48    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   48    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   48    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   48    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   48    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   48    0         0        
AT+CGNSSPWR?

+CGNSSPWR: 1,**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   57    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   57    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   57    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   57    0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   179   0         0        
AT+CGNSSPWR?

(Blocks of asterisks continued below)
**********************************************
GPS_BuiltIn shows:
Start modem...
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT
AT
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: INCORPORATED
Model: A7608SA-H
Revision: A7600M7_V5.0
IMEI: 860371050882459
+GCAP: +CGSM,+FCLASS,+DS

OK
[10946] ### Modem: A7608SA-H
Model Name:A7608SA-H
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7608SA-H
Revision: A50C4B04A7600M7
A7600M7_B04V05_220121
QCN:
IMEI: --REMOVED--
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
Enabling GPS/GNSS/GLONASS
AT+CVAUXS=1

OK
AT+CGNSSPWR=1

OK

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CPIN: READY

SMS DONE

PB DONE

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGEV: EPS PDN ACT 1

+NITZ: PLMN Long Name: Telstra, 

+NITZ: PLMN Short Name: Telstra, 

+CGEV: ME PDN ACT 8,0

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
(Repeats below)

Viuytgrfe, 2025-02-01:
Have the same issue for almost 3 months and i haven't found an answer. The only way to fix it is to heat up the connection between antenna and the board, but it helps for like 30 minutes. So, i'd like to get an answer too!

lewisxhe, 2025-02-06:
If the RX count of GPS_NMEA_Parse is always increasing, then it can be determined that there is no problem with the software and the GPS is outputting NMEA sentences normally. The only thing that needs to be confirmed is whether the GPS antenna base is powered by an antenna voltage of about 3.0V. If there is no power supply, the GPS cannot search for signals.

closed 2025-03-24 by github-actions[bot]

Akebs, 2025-04-15:
I was having the same problems with A7608E-H S3. Changed GPS Baud from115200 to 9600, and there was light.  Changing GPS mode from 3 to 4 helped improving positioning in my case .

HUGProgra, 2025-08-20:
/**
 * @file      GPS_BuiltIn.ino
 * @author    Lewis He (lewishe@outlook.com)
 * @license   MIT
 * @copyright Copyright (c) 2023  Shenzhen Xin Yuan Electronic Technology Co., Ltd
 * @date      2023-06-28
 *
 */

#define UART_BAUD 115200
#define MODEM_DTR_PIN 7
#define MODEM_TXD_PIN 17
#define MODEM_RXD_PIN 18
#define MODEM_PWR_PIN 15
#define BAT_ADC 4
#define BOARD_POWER_ON 12
#define MODEM_RI_PIN 33
#define MODEM_RST_PIN 5
#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13

#define TINY_GSM_MODEM_SIM7600   //A7608 Compatible with SIM7600 AT instructions
#define TINY_GSM_RX_BUFFER 1024  // Set RX buffer to 1Kb

#define SerialAT Serial1
// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

bool streamGetLength(char *buf, int8_t numChars,
                     const uint32_t timeout_ms = 1000L) {
  if (!buf) {
    return false;
  }

  int8_t numCharsReady = -1;
  uint32_t startMillis = millis();
  while (millis() - startMillis < timeout_ms && (numCharsReady = modem.stream.available()) < numChars) {
    TINY_GSM_YIELD();
  }

  if (numCharsReady >= numChars) {
    modem.stream.readBytes(buf, numChars);
    return true;
  }

  return false;
}

int16_t streamGetIntLength(int8_t numChars,
                           const uint32_t timeout_ms = 1000L) {
  char buf[numChars + 1];
  if (streamGetLength(buf, numChars, timeout_ms)) {
[... 393 lines cut ...]
  unsigned long timeout = millis() + 30000;  // เพิ่มเวลา timeout เป็น 30 วินาที
  bool gpsReady = false;
  while (millis() < timeout) {
    if (modem.stream.available()) {
      String response = modem.stream.readStringUntil('\n');
      response.trim();
      Serial.println(response);  // แสดงผลการตอบกลับที่ได้รับ
      if (response.equals("+CGNSSPWR: READY!")) {
        gpsReady = true;
        break;
      }
    }
    delay(100);  // หน่วงเวลาเล็กน้อยเพื่อไม่ให้ใช้ CPU มากเกินไป
  }

  if (gpsReady) {
    Serial.println("GPS Ready!");
  } else {
    Serial.println("GPS Timeout!");
  }
}
void loop() {
  getGNSS();
  delay(1000);
}

HUGProgra, 2025-08-20:
ผมใช้ ESP32-S3 A7806E-H และใช้โค้ดด้านบนแล้วและมันได้ผล สามารถอ่านตำแหน่งจาก GPS ได้มากถึง 15 satellite