Xinyuan-LilyGO/LilyGo-Modem-Series #220: T-A7670 powered by battery does not power up

r3v1, 2025-02-01:
Hi, 

Like other comments and posts related to problems with starting the module via battery and after trying all the possible solutions discussed in the other issues, none of them worked.

I have read the documentation, analysed the schematic, set pin 12 high at startup but nothing is effective.

Background:

- Lilygo version: LilyGO-T-A7670 v1.2 (2022-7-18)
- IDE: ESP-IDF v5.3.2

Minimum working example (extracted from bigger project where everything works perfectly via USB power):
- Connected via USB, modem status LEDs (red) and module LED (blue) light up correctly .
- Connected via DC power supply (emulating battery) with 4V and max 2.5A: nothing lights up.

```c
#include "freertos/FreeRTOS.h"
#include "esp_log.h"
#include "driver/gpio.h"
static const char *MAIN_TAG = "main";

#define MODEM_BAUDRATE 115200
#define MODEM_DTR_PIN 25
#define MODEM_TX_PIN 26
#define MODEM_RX_PIN 27
#define BOARD_PWRKEY_PIN 4
#define BOARD_ADC_PIN 35
#define BOARD_POWERON_PIN 12
#define MODEM_RESET_PIN 5
#define MODEM_RESET_LEVEL 1

void config_modem_gpios()
{
    // Turn on DC boost to power on the modem
    gpio_set_direction(BOARD_POWERON_PIN, GPIO_MODE_OUTPUT);
    gpio_set_level(BOARD_POWERON_PIN, 1);
}

void power_on_modem()
{
    ESP_LOGI(MAIN_TAG, "Power on the modem");

#ifdef MODEM_RESET_PIN
    // Set modem reset pin ,reset modem
    gpio_set_direction(MODEM_RESET_PIN, GPIO_MODE_OUTPUT);
    gpio_set_level(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
    vTaskDelay(pdMS_TO_TICKS(100));
    gpio_set_level(MODEM_RESET_PIN, MODEM_RESET_LEVEL);
    vTaskDelay(pdMS_TO_TICKS(2600));
    gpio_set_level(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
#endif
    gpio_set_direction(BOARD_PWRKEY_PIN, GPIO_MODE_OUTPUT);
    gpio_set_level(BOARD_PWRKEY_PIN, 0);
    vTaskDelay(pdMS_TO_TICKS(100));
    gpio_set_level(BOARD_PWRKEY_PIN, 1);
    vTaskDelay(pdMS_TO_TICKS(100));
    gpio_set_level(BOARD_PWRKEY_PIN, 0);
}

void app_main(void)
{
    config_modem_gpios();
    power_on_modem();
}
```

Other debug info:
- VBAT pin voltage: ~4.2V when USB connected and ~4V with 4V DC power supply.
- IO12 does not seem to be HIGH with any power supply despite the script. 
- PWRKEY pin voltage: ~3.7V  when USB connected and ~0V with 4V DC power supply.
- Blue LED does not light up with DC power supply.
- ESP32 EN and 3V3 pins: ~3.3V with both supplies.
- Modem's STATUS LED blinks for little miliseconds once when connected to DC power supply.

Any suggestion on this?

lewisxhe, 2025-02-06:
gpio_set_direction does not initialize GPIO correctly. See the following modification
```c

#include "freertos/FreeRTOS.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "esp_timer.h"
#include <string.h>

static const char *MAIN_TAG = "main";

#define MODEM_BAUDRATE 115200
#define MODEM_DTR_PIN 25
#define MODEM_TX_PIN 26
#define MODEM_RX_PIN 27
#define BOARD_PWRKEY_PIN 4
#define BOARD_ADC_PIN 35
#define BOARD_POWERON_PIN 12
#define MODEM_RESET_PIN 5
#define MODEM_RESET_LEVEL 1

#define INPUT             0x01
#define OUTPUT            0x03 

void pinMode(uint32_t gpio, uint8_t mode)
{
    gpio_config_t config;
    memset(&config, 0, sizeof(config));
    config.pin_bit_mask = 1ULL << gpio;
    switch (mode) {
    case INPUT:
        config.mode = GPIO_MODE_INPUT;
        break;
    case OUTPUT:
        config.mode = GPIO_MODE_OUTPUT;
        break;
    }
    config.pull_up_en = GPIO_PULLUP_DISABLE;
    config.pull_down_en = GPIO_PULLDOWN_DISABLE;
    config.intr_type = GPIO_INTR_DISABLE;
    ESP_ERROR_CHECK(gpio_config(&config));
}

void digitalWrite(uint32_t gpio, uint8_t level)
{
    gpio_set_level((gpio_num_t )gpio, level);
}

void config_modem_gpios()
{
    // Turn on DC boost to power on the modem
    // gpio_set_direction(BOARD_POWERON_PIN, GPIO_MODE_OUTPUT);
    pinMode(BOARD_POWERON_PIN,OUTPUT);
    gpio_set_level(BOARD_POWERON_PIN, 1);
}

void power_on_modem()
{
    ESP_LOGI(MAIN_TAG, "Power on the modem");

#ifdef MODEM_RESET_PIN
    // Set modem reset pin ,reset modem
    // gpio_set_direction(MODEM_RESET_PIN, GPIO_MODE_OUTPUT);
    pinMode(MODEM_RESET_PIN,OUTPUT);
    gpio_set_level(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
    vTaskDelay(pdMS_TO_TICKS(100));
    gpio_set_level(MODEM_RESET_PIN, MODEM_RESET_LEVEL);
    vTaskDelay(pdMS_TO_TICKS(2600));
    gpio_set_level(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
#endif
    // gpio_set_direction(BOARD_PWRKEY_PIN, GPIO_MODE_OUTPUT);
    pinMode(BOARD_PWRKEY_PIN,OUTPUT);
    gpio_set_level(BOARD_PWRKEY_PIN, 0);
    vTaskDelay(pdMS_TO_TICKS(100));
    gpio_set_level(BOARD_PWRKEY_PIN, 1);
    vTaskDelay(pdMS_TO_TICKS(100));
    gpio_set_level(BOARD_PWRKEY_PIN, 0);

}

void app_main(void)
{
    config_modem_gpios();
    power_on_modem();
}
```

r3v1, 2025-02-06:
Thanks, it works!

closed 2025-02-06 by r3v1