Xinyuan-LilyGO/LilyGo-Modem-Series #221: Problem with the GSM network T Call A7670E

imboqqentt, 2025-02-03:
[X] I have read the troubleshooting section of the README

## What type of issue is this?

[ ] Error or problem compiling the library
[ ] Question or request for help
[ ] Hardware issue
[X] Network issue
[ ] GPS issue

### What are you using?

Modem: 
Modem: 

LilyGO T-Call A7670E

[ ] T-A7670X-ESP32
[X] T-Call-A7670X-V1-0-ESP32
[ ] T-SIM7672G-ESP32S3
[ ] T-A7608X-ESP32
[ ] T-A7608X-S3-ESP32
[ ] T-PCIE-A7670X-ESP32

Examples: 

examples/ATdebug

### AT+SIMCOMATI 

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
QCN:
IMEI: 862771076824922
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK

### Scenario, steps to reproduce

Im trying to connect my board to the GSM local network.

### Expected Results

i want to connect to the GSM local network, but when i use te AT+CREG? command shows this:

AT+CREG?

+CREG: 0,6

AT+COPS?

+COPS: 0,2,"73001",7

AT+CNMP?

+CNMP: 38

AT+CSQ

+CSQ: 26,99

### Debug and AT Command Logs

lewisxhe, 2025-02-06:
Check whether your APN is configured correctly or whether the SIM card is available. This error indicates that there is a problem with the SIM card.
[image]

imboqqentt, 2025-02-06:
Bro, i guess that the problem isnt the simcard, because im using my own simcard, the simcard of my cellphone. 
Another idea?
im trying to use the atdebug example...

when i use the command 
AT+CNMP=2

the board register on the network, but i cant do voicecalls or send SMS, i dont know how to proceed pls help

OK
AT+CREG?

+CREG: 0,1

OK
AT+CNMP?

+CNMP: 2

OK

AT+CSQ

+CSQ: 25,99

OK
AT+COPS?

+COPS: 0,2,"73001",7

OK
ATD+56965334355;

OK

+CLCC: 1,0,2,0,0,"+56965334355",145,"",,0

+CLCC: 1,0,6,0,0,"+56965334355",145,"",,0

NO CARRIER

VOICE CALL: END

As you can see, i cant do voicecalls since the board to a cellphone, and when i try to call the board is the same problem, maybe that can give you a hint.

Regards!

lewisxhe, 2025-02-07:
Send AT+CPSI? The current network registration status will be returned. First check which network it is registered in. Also, pay attention to the messages reported by the modem when turning on the phone. Are there any PD DONE and SMS DONE?

lewisxhe, 2025-02-08:
I don't see any problem. Please follow the documentation to upgrade the modem to the latest firmware, then capture the modem log. After starting to capture, try to send AT commands to make a phone call. Then send the log here. I need SIMCOM's assistance.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/a7670x_capture_log.md

imboqqentt, 2025-02-17:
Hello man, i tried to use the board in multiple times, but is the same, the problem isnt solve, i had to change the board for a A7608E-H and all works correctly, i dont know if the problem is only mine or is something general in this boards,  but im not happy with my purchase.

lewisxhe, 2025-02-18:
Is it OK to replace it with another T-A7608?

llussa, 2025-03-21:
hello, I have also problems with gsm network.
i have a program that read some temperarute and humidity and send it to blynk.
It was possible to do it with Wifi, but I have problems with GSM, here are the serial monitor text :
5:35:50.235 -> Iniciant sistema...
15:36:08.442 -> Consultant firmware del mòdem...
15:36:09.415 -> Versió firmware: +CPIN: READY
15:36:09.457 -> Versió firmware: SMS DONE
15:36:09.457 -> Versió firmware: +CGEV: EPS PDN ACT 1
15:36:09.458 -> Versió firmware: +CGEV: ME PDN ACT 8,0
15:36:09.458 -> Versió firmware: PB DONE
15:36:09.458 -> Versió firmware: +CGMR: A110B01A7670M7_F
15:36:09.458 -> Versió firmware: OK
15:36:39.481 -> [50282] ### Unhandled: *ATREADY: 1
15:36:39.481 -> 
15:36:39.481 -> +CPIN: READY
15:36:39.481 -> +CGEV: EPS PDN ACT 1
15:36:39.481 -> SMS DONE
15:36:39.481 -> +CGEV: ME PDN ACT 8,0
15:36:39.481 -> PB DONE
15:36:39.481 -> [50282] ### TinyGSM Version: 1.0.0
15:36:39.481 -> [50292] ### TinyGSM Compiled Module:  TinyGsmClientA7670
15:36:39.552 -> [50328] ### Modem: A7670E-FASE
15:36:39.552 -> [50328] ### Modem: A7670E-FASE
15:36:39.742 -> [50574] ### TinyGSM Version: 1.0.0
15:36:39.742 -> [50574] ### TinyGSM Compiled Module:  TinyGsmClientA7670
15:36:39.777 -> [50602] ### Modem: A7670E-FASE
15:36:39.777 -> [50602] ### Modem: A7670E-FASE
15:39:47.673 -> [238493] ### Unhandled: +CIPCLOSE: 0,0
15:42:43.006 -> ** CONNECTAT A BLYNK **
15:42:43.169 -> Temp: 21.67°C, Hum: 77.06
15:43:04.275 -> [435088] ### Closed:  0
15:44:03.311 -> Temp: 21.72°C, Hum: 77.03
15:45:24.305 -> Temp: 21.77°C, Hum: 76.93
15:48:07.271 -> Temp: 21.74°C, Hum: 75.83
15:49:07.398 -> Temp: 21.92°C, Hum: 76.29
15:49:08.767 -> [799587] ### Closed:  0

closed 2025-05-05 by github-actions[bot]

gcandrade10, 2025-05-05:
Hi! I'm having the same issue:

AT+CREG? 
+CREG: 0,6

with 

AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B03SIM767XM5A_M
SIM767XM5_B03V02_240914
IMEI: 864643060159763

please help