Xinyuan-LilyGO/LilyGo-Modem-Series #222: Separate UART for GNSS NMEA - T-SIM7670G S3

JENebel, 2025-02-03:
It is possible to redirect the GNSS NMEA output to the 7670G's usb NMEA port with
`AT+CGNSSPORTSWITCH=0,0`
But are the relevant pins connected anywhere, so I can access them from the ESP32?

My goal is to have continous GNSS output, and also upload data via GSM at the same time. So it would be very nice to have the NMEA output separated physically from the AT command responses.
The alternative would be to separate them in software, but this seems fragile and cumbersome.

lewisxhe, 2025-02-06:
The current hardware is not capable of this. It can be seen that it does have GPS TX and RX, but this has been connected to the modem's RX TX.
[image]

JENebel, 2025-02-09:
Alright, thank you. I managed to filter the messages in software instead

closed 2025-02-09 by JENebel