Xinyuan-LilyGO/LilyGo-Modem-Series #224: LILYGO T-A7670G network example doesn´t work

davidarbelaez, 2025-02-05:
Hello, I have a trouble with network.ino example, I am new to Lilygo 4G development boards.

First the code network.ino got an error that said: "#error "Please define GSM modem model", so I defined modem model at the beggining of the code as follows: #define TINY_GSM_MODEM_A7670 - is it right?. Then I got another error that says "SerialAT is not declared in this scope". It says "do you mean Serial2?". Should I change it for Serial2? 

¿What sould I do?

Thank you very much

lewisxhe, 2025-02-06:
Have you read the README ?
[image]

davidarbelaez, 2025-02-06:
Great I did it, and now it works. Thanks. Another question, have you linked a Lilygo board to Arduino Cloud platform? If so, how can I do it?

lewisxhe, 2025-02-07:
Please look at the protocol of the platform you want to connect to and view the relevant documents

closed 2025-03-23 by github-actions[bot]

jawwadcreation, 2025-04-20:
I recently purchased Liliygo A7670G R2 with external GPS 
but can't connect with GPS I try many things but not working, 

**please give some code example for GPS start working, also I try many AT commands but all of those commands return Error.**

see this code I have to try at last:

#include <TinyGPS++.h>
#include <HardwareSerial.h>

TinyGPSPlus gps;
HardwareSerial gpsSerial(1);  // UART1 for GPS

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("🛰️ GPS Direct UART Reading Test...");

  // 🔁 One line at a time active rakho, baaki comment karo

  gpsSerial.begin(9600, SERIAL_8N1, 21, -1);  // Try GPIO21
  // gpsSerial.begin(9600, SERIAL_8N1, 16, -1);  // Try GPIO16
  // gpsSerial.begin(9600, SERIAL_8N1, 33, -1);  // Try GPIO33
  // gpsSerial.begin(9600, SERIAL_8N1, 36, -1);  // Try GPIO36 (input only)
}

void loop() {
  while (gpsSerial.available()) {
    gps.encode(gpsSerial.read());
  }

  if (gps.location.isUpdated()) {
    Serial.println("📍 GPS FIX RECEIVED!");
    Serial.print("Latitude: ");
    Serial.println(gps.location.lat(), 6);
    Serial.print("Longitude: ");
    Serial.println(gps.location.lng(), 6);
    Serial.print("Satellites: ");
    Serial.println(gps.satellites.value());
    Serial.println("--------------------");
  }

  delay(1000);
}