Xinyuan-LilyGO/LilyGo-Modem-Series #225: LILYGO T-SIM7670G-S3 ESP32-S3 4G battery charging problem

PIERLUIGI1983, 2025-02-06:
By connecting a 3.5W panel with 5V output, the LILYGO T-SIM7670G-S3 ESP32-S3 4G card does not charge the battery. Do I need to load a particular code or solder some pins?

lewisxhe, 2025-02-06:
3.5W panel? Do you mean solar panel? ? ? 5V output voltage, 3.5W power solar panel?
No need to load specific code, by default, just connect the solar panel to the solar input terminal, it can be automatically charged

majuss, 2025-02-10:
[image]

My T-A7608-S3 is also not charging from USB.

lewisxhe, 2025-02-12:
@majuss Sorry, we have detected a defect in this batch of diodes. You need to replace the diode in the picture below with a B5817WS SOD-323 package, or other diodes with the same package can be tested.
If you do not use the solar panel and USB at the same time, short-circuiting this diode can solve this problem.

majuss, 2025-02-12:
This is somewhat of a critical error. I already built a fallback method and now I had to remove everything... Is there some kind of compensation for this? I lost at least 8 hours of my time. Why didn't you email all customers that bought this board?

lewisxhe, 2025-02-13:
@majuss  Please contact your dealer.

closed 2025-03-29 by github-actions[bot]