Xinyuan-LilyGO/LilyGo-Modem-Series #229: SD card failure of battery power

I-em-zain, 2025-02-13:
My SD card is being detected on USB power but not on battery power. Anyone else faced this issue?

lewisxhe, 2025-02-13:
Is IO 12 set to HIGH?

I-em-zain, 2025-02-14:
> Is IO 12 set to HIGH?

Yes, in GSM.h
#define BOARD_POWERON_PIN (12)

#ifdef BOARD_POWERON_PIN
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif

Moreover can you please share schematics for A7670 v1.2

lewisxhe, 2025-02-14:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/schematic/T-A7670X-V1.1.pdf 

Same as 1.2

lewisxhe, 2025-02-14:
Is there a multimeter that can directly measure whether the SD pin is powered?
[image]

I-em-zain, 2025-02-14:
Vbus on USB power is at around 5 V and 2.2 mV on battery

lewisxhe, 2025-02-15:
Incorrect, which position did you measure? If the USB-powered SD card can be powered, then Q11 should be turned on. If the battery is powered and GPIO12 is set to a high level, then Q11 should also be turned on.
Connect the battery, set GPIO12 to a high level, and then measure the test voltage of the two test points in the figure below, with the black test lead placed on GND

[image]

closed 2025-04-01 by github-actions[bot]