Xinyuan-LilyGO/LilyGo-Modem-Series #231: A7672S not connecting with network

Nikitanagar, 2025-02-20:
Hello,
I am working on the SIMCom A7672S LTE module and trying to connect it to the Jio (405861) network in India, but it is failing to register. Below are the details of my setup and the tested AT commands.

Module Details:

Model: A7672S
Network: Jio 
SIM Status: Ready (AT+CPIN? → +CPIN: READY)
Signal Strength: Good (AT+CSQ → +CSQ: 31,99)
Current Network Mode: Auto (AT+CNMP? → +CNMP: 2)
Frequency Bands: 
LTE-TDD Supported: B34/B38/B39/B40/B41 ,
LTE-FDD Supported: B1/B3/B5/B8,
GSM: GSM/GPRS/EDGE 900/1800 MHz

Tested Commands & Responses:

```
AT+CREG?
+CREG: 1,0
OK

AT+COPS=?
+COPS: (1,"Vodafone IN","Vodafone","40486",0),
       (1,"CellOne","CellOne","40471",0),
       (1,"IND airtel","airtel","40445",0),
       (1,"Vodafone IN","Vodafone","40486",7),
       (1,"IND-JIO","JIO","405861",7),
       (1,"IND airtel","airtel","40445",7),
       (0,1,2,3,4),(0,1,2)
OK

AT+COPS=1,2,"405861",7
+CME ERROR: PLMN not allowed (#11)

AT+CNBP=0X0000000000000000,0X0000000080000014,0X0000
OK

AT+CFUN=1,1
OK
+CGEV: ME DETACH

AT+CREG?
+CREG: 0,0
OK

+CREG: 3
+CREG: 0
+CME ERROR: phone failure
```

Issue:
When attempting to manually select the Jio network using AT+COPS= ******, I receive the following errors:

1️⃣ PLMN Not Allowed (+CME ERROR: PLMN not allowed (11))
2️⃣ Phone Failure (+CME ERROR: phone failure)

Additionally, AT+CREG? returns +CREG: 1,0 (registered but not roaming), and sometimes +CREG: 3 (registration denied).

Steps Taken So Far:
 1. Ensured SIM is active and works in a phone
 2. Verified Jio LTE network is available using AT+COPS=?
 3. Set supported LTE bands using 
    AT+CNBP=0X0000000000000000,0X0000000080000014,0X0000
 4. Restarted the module (AT+CFUN=1,1)
 5. Tried forcing LTE mode (AT+CNMP=38).

Any insights or suggestions would be greatly appreciated. Thanks in advance!

lewisxhe, 2025-02-20:
AT+CNBP Do not set this lightly, it is a dangerous operation

Nikitanagar, 2025-02-20:
yes I set the APN using AT+CGDCONT=1,"IP","JioNet".

lewisxhe, 2025-02-20:
You said that sometimes it can return to "registered", but sometimes it returns "registration rejected", is that right? This should be a problem with the SIM card.

lewisxhe, 2025-02-20:
Are you using 7670 or 7672?

SIM7672 does not have the CNMP command

Nikitanagar, 2025-02-20:
1. I am getting this AT+CREG?
+CREG: 1,3  when I am setting the AT+CREG = 1.  
2.  SIM Card is working fine in mobile .
3.  I am using SIMCOM A7672S

closed 2025-04-06 by github-actions[bot]