Xinyuan-LilyGO/LilyGo-Modem-Series #232: Major Issue with Lilygo-T-A7670 on Battery: Everytime "Brownout detector was triggered" once connected on battery and devices get hanged and need manual reset

farhanahmad-ese, 2025-02-24:
Dear Lilygo Team and Developers,

There is major issues in this board. Whenever tries to connect it with battery Li-Ion or Lipo every time it output "Brownout detector was triggered". I have disable brownout detection but our devices get failed/hanged on this stage. There are lot of devices we have procured and whole project has been messed up. When we have connected FTDI adapter to save log, this is the state when device get hangs anytime and we have to go for manual reset. There is lot of unreliability with this product. This log is just got by programming your own example MQTTClient without doing any change just connecting the module on battery. 

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4

Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start modem...
[6837] Initializing modem...
[6837] ### TinyGSM Version: 1.0.0
[6837] ### TinyGSM Compiled Module:  TinyGsmClientA7670
[6897] ### Modem: A7670E-FASE
[6897] ### Modem: A7670E-FASE

lewisxhe, 2025-02-24:
There is another problem, POWER ON (GPIO12) must be turned on when power is on, otherwise there is a chance of triggering a power-off reset.

farhanahmad-ese, 2025-02-24:
Yes. The batteries are reliable. Yes, I have tried to power the board using USB-C connected through lipo battery charging circuit and the same battery was being used. 
The problem occurs when you GPIO 12 is turned ON.

  // Turn on DC boost to power on the modem
```
#ifdef BOARD_POWERON_PIN
  pinMode(BOARD_POWERON_PIN, OUTPUT);
  digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif
```
This is the part of the code where the Brownout Detector Triggers.

lewisxhe, 2025-02-25:
Did you enable GPIO12 at the very beginning of the setup function?

If you enable it after setup, it may cause power failure!

farhanahmad-ese, 2025-02-25:
The Serial Log I have shared is from your examples "MQTT Client".

[image]

And facing issues. There is an issue with hardware. I am not sure if that could be resolve by using a capacitor to stabilized the voltage. 

The hardware designer should not use GPIO 12 as this is strapping pin. He could have used any other GPIO.

farhanahmad-ese, 2025-02-26:
Here is the battery voltage waveform captured using oscilloscope. The battery voltage is 4.0 and when modem tried to power up then it drop below 3.0 V

[image]

lewisxhe, 2025-02-26:
This problem is a known issue and can only be solved by calling GOIO12 to set the high level at the beginning of setup. This problem has been discussed in another post.

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/85

farhanahmad-ese, 2025-02-26:
This problem is not solved. We have lot of devices now in field and they are getting hanged. It's fault of hardware designer who created mess for everyone who bought this module. We cannot afford high current consumption as devices are battery operated and remotely deployed.
It was your code where GPIO 12 was set high at beginning of setup but still exists.

lewisxhe, 2025-02-27:
If that's the case, then there may really be something wrong with the board you have. How many did you buy?

farhanahmad-ese, 2025-02-28:
Dear Lewisxhe,

We have 55 Lilygo T-7670A devices as pilot project. Just for information of Lilygo team, that we have Lilygo T3 S3 LoRaWAN based device that is under testing and we have to deploy 500 nodes as pilot. Later on, we have to install 1100 nodes at second stage. 
Currently, we are losing trust on Lilygo devices and if Lilygo is not serious to solve the issue then there will be other options.

lewisxhe, 2025-03-03:
55 boards all called GPIO12 SET HIGH during startup, which triggered ESP undervoltage reset? ?

farhanahmad-ese, 2025-03-03:
I cannot check all board because most of them has been deployed in field. I have called GPIO12 HIGH during startup as the code snippet is given below:
```
void setup()
{
    // Turn on DC boost to power on the modem
#ifdef BOARD_POWERON_PIN
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif
    // WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
    Serial.begin(115200); // Start Serial communication for debugging
    while (!Serial)
    {
    } // Wait for Serial to be ready
    LOG_MESSAGE(millis(), "Serial Started at 115200kbps", "✔");
```
Furthermore, I have disabled brownout detection which ultimately resulted in `ESP_RST_INT_WDT` reset reason. Thus, nothing has worked. 
I have 08 board with me in lab and they are problematic. 
We have Lilygo T A7670E V1.2 2022-7-18. I have seen issue # 85 where you have suggested to remove a MOS. Now we want to try on our board. I am sharing picture of our board, please suggest which MOS to remove and which pins to short. 

Furthermore, we have installed a ATTiny85 microcontroller with Lilygo module which resets Lilygo after specified interval as we do not have any other options to manually reset the devices in field.

[image]

lewisxhe, 2025-03-04:
Please see the picture

[image]

Can you test this code by placing it in setup and then executing it later to keep the modem turned on immediately when booting?

[image]

farhanahmad-ese, 2025-03-04:
Yes. I have removed the mentioned MOS and kept GPIO12 HIGH at start of void setup(). However, the current consumption reaches 600uA. 

[image]

For now, Brownout detection is not triggering. However, we need to test if it gets hanged after long operation such as for days and weeks. Like like mentioned in Issue # [105](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/105)

lewisxhe, 2025-03-05:
OK. Waiting for your test results

farhanahmad-ese, 2025-03-22:
I came to see the following image from this github repository. Could you please explain in which version and why it's necessary to remove the mention resistor before going into sleep mode

[image]

lewisxhe, 2025-03-24:
This should be the VBUS voltage input of the modem

[image]

farhanahmad-ese, 2025-03-24:
We have v1.2 and here is the picture of module. Do we have to remove this resistor and if not remove then what will be the effect? I have noticed that it "000" printed on this

[image]

lewisxhe, 2025-03-24:
According to the schematic diagram, if this resistor is not removed, the MODEM VBUS will not be powered.

farhanahmad-ese, 2025-04-28:
Dear Lewisxhe,
We have faced this issued in Lilygo T-A7670 v1.2. Now we want to procure more devices. Has Lilygo solved this hardware issue like removing MOSFET and produced new version?

lewisxhe, 2025-04-29:
No, the current and old versions have not been revised. A more stable A7670 S3 version will be released later. It fixes many known issues and is the main recommended version. If you are not in a hurry, you can wait.

bsfconception, 2025-05-12:
Hi @farhanahmad-ese 
Do you confirm that removing the MOS and adding the shortcut solves the issue ?
I'm a bit upset about LilyGO designs ... fes years ago I did encounter some issue with very badly designed modules based on SIM800.
I did hope that this one was better, but no way.
I'll "correct" my module and return to design my own ESP32-A7670 solution ....
Thank for this investigation that saves me a lot to time

farhanahmad-ese, 2025-05-12:
Dear @bsfconception,
Yes. Previously, the device hanged after couple of day and now after removing MOSFET its resolved. However, the current consumption increased from 146uA to 600uA. 
Yes, Lilygo has still not released version after v1.2 in which these issues would be fixed. There is another alternate by Waveshare ESP32-S3 which has same feature as Lilygo T-A7670E and we are going to test that as well. 
[https://www.waveshare.com/esp32-s3-a7670e-4g.htm](url)

bsfconception, 2025-05-12:
Thanks for the link about the alternative solution.
I'm currently using a A7270 PCB from "And Global" with home made ESP32 PCB. 
I did expect to use this PCB, but I'm returning to my solution ;-)
Furthermore, the Lylogo and waveshare are a bit too large for my usage
kind regards

deeja, 2025-05-19:
I don't know why it's not mentioned anywhere, but I've added a 2200uF low ESR Electrolytic capacitor to mine and I'm seeing SIGNIFICANTLY improved results. I don't know what the ideal cap size is, but this has changed everything. 
Running at 3.4 v without issues, although it's recommended to be 3.5+ at minimum.

https://www.jaycar.co.nz/2200uf-10vdc-low-esr-electrolytic-capacitor/p/RE6306 
[image]

deeja, 2025-05-20:
I've added this as an issue https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/287

farhanahmad-ese, 2025-05-20:
> Thanks for the link about the alternative solution. I'm currently using a A7270 PCB from "And Global" with home made ESP32 PCB. I did expect to use this PCB, but I'm returning to my solution ;-) Furthermore, the Lylogo and waveshare are a bit too large for my usage kind regards

Dear @bsfconception 
Sorry for late response. I was extremely busy. Thank you for sharing valuable information. Actually, we need ESP32 + GSM + card reader + BMS + solar in single module. That's why we have chosen these two modules.

bsfconception, 2025-05-20:
> I don't know why it's not mentioned anywhere, but I've added a 2200uF low ESR Electrolytic capacitor to mine and I'm seeing SIGNIFICANTLY improved results. I don't know what the ideal cap size is, [...]

What a huge capacitor. I did think about a simular solution, because the actual capacitors are a bit small. 
On my own desigined system, base on ESP32 Wrower and A7270, I did use seveval 470uF tantale capacitor for 3.3V and A7075 power supply. 
But, looking again about the schematic, I'm less confitend on the product. The management of the power supply switches, the LC power filters and unstable A7670 supply at statup are really concerning me.
I'll shortcut the A7670 3.3V control to have it always ON and try to use rather that put the module aside.
But I definitively return to my solution (10 GPS tracking systems running perfectly for several years). I did expect to save time and money, but I'll continue to solder ;-)

deeja, 2025-05-20:
@bsfconception - on some of the Lora IoT batteries they are even bigger but only because Li-SOCl2 cells are only capable of a small current draw. I hope that this setup will help with old battery ESR, and it looks like it is helping. Also should mitigate battery degradation from high current bursts. 
https://www.lobaro.com/en/products/single-3v6-iot-battery/
[image]

GTorreil, 2025-05-29:
Hello, 

I also have devices based on this board that freeze after some time in the wild (typically 10-20 days). I'm using WDT so I can rule out a software hang. Also, when the devices are frozen there is no output on the serial port. I'm setting GPIO 12 high at the beginning of setup. It seems related to this issue. Modifying the board is not an option for me, neither increasing power draw. It's been happening for months, I've tried many many changes. Can I reset the board from time to time from the software to fix it ?

farhanahmad-ese, 2025-05-30:
> Hello,
>
> I also have devices based on this board that freeze after some time in the wild (typically 10-20 days). I'm using WDT so I can rule out a software hang. Also, when the devices are frozen [...]

We were using WDT to reset in case of software hang, however, the issues was not resolved. The device still get hanged and we had to manually reset. 

Now we have programmed ATTiny85 as external hard reset switch or it can be like external Watch Dog Timer. Our devices push telemetry after every 20 minutes. We have programmed Lilygo such that i wake up ATTiny85 and push telemetry and ATTiny85 goes back to sleep for next 30 minutes. If Lilygo gets hang then ATTiny85 will wake up after 30 minutes and hardware reset device throught 2N7000 MOSFET. This approach is working perfectly fine. Although this is not standard approach, however, we have remote deployments and cannot hard reset if device hang.

Previously, we had our customer PCB board for GSM based sensor nodes, however, there were production limitation for bulk orders. That's why moved to Lilygo board. Let's wait if they fix this major issue. Nevertherless, removing the mentioned MOSFET increased sleep current from 146uA to 600uA which is another challenge. Currently, energy is being harvested through solar, however, we are working to optimize this.

One proposed approach is using TPL5110 Nano Power System Timer for Power Gating. This operates at 35 nA and shutdown your main micrcontroller to save lot of energy and promises long battery life. This is standard approach in various sensor nodes.

maxibrunialti, 2025-07-03:
> We were using WDT to reset in case of software hang, however, the issues was not resolved. The device still get hanged and we had to manually reset.
>
> Now we have programmed ATTiny85 as external hard [...]

Dear colleague,

I’m following your case closely as I’m also experiencing power issues with my devices.
They work correctly as long as they’re connected through a USB-C charger.
The problem arises when we connect them to the car’s power supply, which is where our client needs to use them.
The device freezes and there’s no way to recover it.
However, if I remove the battery and power it only through the car’s power source, the problem disappears.
Have you seen any improvement after removing the MOS?

@lewisxhe

lewisxhe, 2025-07-04:
Removing the MOS will increase the sleep current a bit, which is described above and will not cause any problems in actual use.