Xinyuan-LilyGO/LilyGo-Modem-Series #233: Modem disconnects when writing to tft

Econut, 2025-02-24:
So i have got my A7670E to work it connects to lte, gps and wireless ok and i have multiple temperature sensors connected and they are all working ok.
I decided to add a 2.2 tft screen spi 240x320 iliy9341

So the screen works ok when it boots up it sends the info i have setup in the setup loop to the screen, but once the modem has finished starting the screen gets an update and as soon as the update starts the modem disconnects and mqtt fails to connect anymore. now i thought it was maybe a conflict on a pin so i have tried various pins etc but no luck, as soon as i add the display update script the modem disconnects.

Im currently using the following pins to connect the screen:

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5  
#define TFT_DC   02 
#define TFT_RST  or 33

as soon as i send anything to the screen the modem disconnects.

I was wondering if anyone else had managed to get a screen to work ok with this card?

thanks

lewisxhe, 2025-02-25:
Do you know which GPIOs the modem on the board you are using is connected to? ? ?
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/f23db9ff0a7c1168138e34549f22858f3702fbac/examples/ATdebug/utilities.h#L39

Econut, 2025-02-25:
Thanks for the reply

i was only using the ifo of f the website pin out pictures 

Wow the modem uses a lot of pins it dont leave many for me to use lol

i can see im using cs on pin 5 and thats the modem reset

i will have a play now  

thankyou

closed 2025-03-24 by Econut