Xinyuan-LilyGO/LilyGo-Modem-Series #234: Board isn't recocniced by Arduino IDE and PC

JulianHeckel, 2025-03-07:
It is my first time programming Microcontrollers and I have worked on my A7670G for a couple of weeks without any troubles, but yesterday it stopped showing up in my IDE. I tried restarting my Pc, different USB-C cables, different USB-C Ports but with no effect. It still gets power from my Pc and runs the latest uploaded Code. 
I appreciate any help

lewisxhe, 2025-03-08:
Is the port not recognized now?

JulianHeckel, 2025-03-08:
No, it doesn't show up anywhere on my Pc or in the IDE

lewisxhe, 2025-03-10:
Shoot a video, steps

1. Open the computer device manager

2. Insert the USB-C cable to the A76XX USB-C port

The above steps prove that the port cannot be recognized

Send the video of the above steps to the dealer and let them provide you with after-sales service.

JulianHeckel, 2025-03-10:
I was referred back to this issue chat after providing pictures to the dealer

JulianHeckel, 2025-03-14:
So there is no solution for my issue?

lewisxhe, 2025-03-14:
Not recognizing the port is a hardware issue and I cannot provide a fix

closed 2025-03-14 by JulianHeckel