Xinyuan-LilyGO/LilyGo-Modem-Series #235: Can't import library to Arduino IDE

Makuu6, 2025-03-11:
Can someone help me assist on adding ZIP file for this library after downloading this GitHub repo to ZIP the Arduino IDE won't install it to it's library , I'm using LILYGO ESP32 S3 SIM7670G 4G Module. Many thanks!

[image]

Makuu6, 2025-03-11:
[image]

lewisxhe, 2025-03-11:
[image]

Makuu6, 2025-03-11:
10:50:14.429 -> Start modem...
10:50:17.449 -> AT
10:50:17.449 -> 
10:50:17.449 -> +CPIN: READY
10:50:17.449 -> 
10:50:17.449 -> SMS DONE
10:50:17.449 -> AT

10:50:17.449 -> OK
10:50:17.449 -> Wait SMS Done.
10:51:57.424 -> Can't wait from sms register ....
10:55:23.556 -> 
10:55:23.556 -> +CMTI: "SM",1

Makuu6, 2025-03-11:
Thank you Ive managed to send SMS ! and receive ! Many Thanks!

closed 2025-04-25 by github-actions[bot]