Xinyuan-LilyGO/LilyGo-Modem-Series #238: A7670E not switching to battery power on USB power failure

mikewhitby, 2025-03-26:
Forgive the potential misunderstanding of what this board should do - I'm very amateur when it comes to ESP32s etc. I was under the impression that given the board has a battery it would be able to survive USB power loss - this is in fact my intended purpose, I want to monitor for when the USB power disappears and send a notification via SMS, the board is intended as a power monitor / alerter.

I've got a simple sketch going and have enabled the battery switch, however when I power on via USB and then pull the USB out the board resets. If I turn the battery switch on the board will boot up and run from battery no problems.

Am I incorrect in saying the board should be able to automatically switching from USB to battery power?

Note that I've disabled brownout protection but this has had no effect.

Thanks!

lewisxhe, 2025-03-27:
Is GPIO 12 set to high level?
[image]

mikewhitby, 2025-03-27:
Sorry I forgot to mention this. I’ve both with and without, same result

lewisxhe, 2025-03-27:
Then please use battery power, turn the switch to ON, and then measure whether there is voltage on the 3.3V pin header.

mikewhitby, 2025-03-27:
Ok, it's 3.29v on that header - I checked by disconnecting usb and switching the battery switch to on - it boots fine and that pin measures a solid 3.29v

mikewhitby, 2025-03-27:
Hmm, I've been playing about a bit more with this - I'm not sure why but it does now seem to be working, I'm disabling brownout detection and it does seem to work.

I do have a 128x64 Pixel I2C SSD1306 display connected so I was assuming that was the issue and went about testing it whilst disconnected and was still having issues, yet now even with the screen connected it seems to be surviving the switch from USB to battery - I honestly can't say what has changed.

It doesn't seem to matter whether I pull pin 12 high or not, is pin 12 only for the modem power?

mikewhitby, 2025-03-27:
Ah ok, so I spoke a bit soon - whilst it does seem to switch over (though not all the time) if I totally yank the USB lead out (instant power loss), it does not work if I simulate a mains power loss scenario by turning off the wall outlet switch where the USB power adaptor is plugged in. I assume this creates a scenario where VBUS drops gradually rather than instantly and the ESP32 does not survive and reboots, in fact sometimes it doesn't even reboot.

lewisxhe, 2025-03-28:
1. Pin 12 is only used to control the power supply of the modem and SD card
2. So your problem now is that when you disconnect the power supply, you don’t always switch back to battery power, right? Have you tested whether the 3.3V of ESP32 has voltage when it is not working?

mikewhitby, 2025-03-28:
Thanks for the clarification on pin 12.

You are correct with point 2, specifically - I have a USB power supply connected to the board, I turn off the wall switch at the power supply to replicate a power failure as accurately as I can (as opposed to pulling out the USB cable which doesn't seem an accurate replication) and the board does one of two things:

1. Just crashes and stops running, I know this as I wrote a sketch which blinks an LED every few seconds and the LED stops blinking (this is by far the most common scenario)
2. Or reboots on battery power (very rarely)

When scenario 1 happens the only way to get the board back working is either:

1. Turn off the USB and battery power, then turn it all back on
2. Press the reset button

I don't have a scope I'm afraid so am relying on my multimeter, but even with that I can see the issue - when the power to the USB adaptor is turned off the voltage dips a lot, I've seen 2.1v, 2.6v etc - I presume I see the variance is simply because the multimeter cannot reflect the change fast enough.

lewisxhe, 2025-04-01:
I want to ask, when the USB power adapter loses power, have you observed whether the ESP can automatically restart? The most reliable method is to connect a USB2TTL to the RX TX of the ESP for log monitoring, or measure whether the 3.3V of the ESP has voltage.

Are you using T-A7670E? The ESP version, the one with 18650, right?

My device will temporarily shut down the ESP after disconnecting the USB, and then restart it through the battery

mikewhitby, 2025-04-01:
> when the USB power adapter loses power, have you observed whether the ESP can automatically restart? 

If I disable brownout detection then it won't restart generally, I presume the voltage drops too low and the ESP32 presumably freezes.

> The most reliable method is to connect a USB2TTL to the RX TX of the ESP for log monitoring

I'll try and set this up

> or measure whether the 3.3V of the ESP has voltage

Well it dips as described before to a level which won't sustain the ESP32

> Are you using T-A7670E? The ESP version, the one with 18650, right?

Yes that one

> My device will temporarily shut down the ESP after disconnecting the USB, and then restart it through the battery

That is what mine is doing, is this as per design? I would have assumed a device like this would work similarly to the Adafruit feather which will continue running on battery without a reboot when USB power is lost

lewisxhe, 2025-04-01:
No! It's not seamless, it loses power, then switches to battery, it loses power in the middle, then switches to battery.

Is that the question? Maybe you understand it as UPS. Is that right?

mikewhitby, 2025-04-01:
> Maybe you understand it as UPS. Is that right?

Absolutely. Other boards like the Feather and FireBeetle ESP32 do this and don't even have the battery holder built in, I would have assumed that a board with a battery and USB power would handle switching of the power source automatically

lewisxhe, 2025-04-02:
OK, that seems to be a misunderstanding. Switching will cause a brief power outage.

closed 2025-05-13 by lewisxhe