Xinyuan-LilyGO/LilyGo-Modem-Series #239: Utilizing BT and wifi commands using AT

Makuu6, 2025-03-30:
Is there any way to test BLE and BT, and WIFI  functionality using my SIM7670G board with AT commands?

lewisxhe, 2025-04-01:
There is no way to use A7670X, the hardware is not reserved, you can use the Bluetooth and Wi-Fi of ESP

Makuu6, 2025-04-01:
ohhh okay thank you! also i cannot determine specifically if these GPIOs are 3.3V or 1.8V i cant be sure when i read the datasheet. do you have any ideas to that?

lewisxhe, 2025-04-02:
Which GPIO are you referring to?

The modem's GPIO is 1.8V

The ESP's GPIO is 3.3V

Makuu6, 2025-04-05:
Is its safe to use RS485 3.3V on ESP GPIOs?

lewisxhe, 2025-04-07:
RS485 transceivers come in two types: 3.3v and 5v. Using 3.3v is safe, while 5v is unsafe.

Makuu6, 2025-04-28:
can you help me again bro once again modem wont show up..just worked yesterday but the next day wont work

lewisxhe, 2025-05-13:
I didn't see your message, is it ok now?

closed 2025-06-26 by github-actions[bot]