Xinyuan-LilyGO/LilyGo-Modem-Series #241: T-A7608E-H Https post Error 715

karrisaiteja, 2025-04-04:
Hi,

I am experiencing an issue with the TTGO-T-A7608E-H devices. I have been using these devices for the past six months, with eight units deployed across multiple locations for data collection. However, since around 7:00 AM IST on March 25, 2024, all the devices have stopped sending data to the cloud. I brought one device back to investigate the issue and encountered error 715, indicating an inability to make HTTPS posts, although MQTT protocols are functioning correctly. I have updated the device firmware and am attaching the device information for your reference.Please assist me in resolving this issue. 

[image]

Thank you in advance.

rikkreeftenberg, 2025-04-05:
My issue #243 seems to be the same. I couldn't "GET" my Azure Function from 28-03-2025. Before that it worked 24/7 for 2 months.

lewisxhe, 2025-04-07:
@karrisaiteja  Are you connecting to Azure too?

karrisaiteja, 2025-04-07:
Yes, I am also connecting to Azure.

lewisxhe, 2025-04-07:
Ok, now I need logs, as mentioned in https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/243.

lewisxhe, 2025-04-10:
The issue has been replied at https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/243, please test

lewisxhe, 2025-04-10:
I also tested it on A7608
```bash
Request URL : https://ikfu.azurewebsites.net/api/GetUtcTime
HTTP Header : HTTP/1.1 200 OK
Content-Type: application/json; charset=utf-8
Date: Thu, 10 Apr 2025 03:05:02 GMT
Server: Kestrel
Transfer-Encoding: chunked
HTTP body : "2025-04-10T03:05:02.7109403Z"
-------------------------------------

» AT+SIMCOMATI
« AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7608SA-H
Revision: A50C4B12A7600M7
A7600M7_B12V01_240315
QCN:
IMEI: 8xxxxxxxxxxxxx
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
```

closed 2025-05-25 by github-actions[bot]