Xinyuan-LilyGO/LilyGo-Modem-Series #243: LilyGO with a SIMCom A7670E suddenly returns errorcode 715 (Handshakefailed) when accessing an Azure Function App

rikkreeftenberg, 2025-04-05:
My  LilyGO TTGO T-SIM-A7670E R2 ESP32 - ESP32-WROVER sudddenly couldn't access my Azure Function App anymore.
It has been functioning non stop for more then 2 months. Without a problem.

And then, out of the blue, it couldn't reach https://ikfu.azurewebsites.net/api/GetUtcTime/ anymore.

I tried the Arduino HttpsBuiltlnGet example program (https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsBuiltlnGet).
An errorcode 715 (Handshakefailed) is returned...

I have tried another web api at another server (https://www.ellerbach.net/DateHeure/) that also returns the time and that respond OK. With a status code 200, and the UTC time.

Below is the console log:

    Network IP:10.8.41.53
    AT+HTTPTERM
    AT+HTTPTERM
    ERROR
    AT+HTTPINIT
    AT+HTTPINIT
    OK
    AT+CSSLCFG="enableSNI",0,1
    AT+CSSLCFG="enableSNI",0,1
    OK
    Request URL : https://ikfu.azurewebsites.net/api/GetUtcTime/
    AT+HTTPPARA="URL","https://ikfu.azurewebsites.net/api/GetUtcTime/"
    AT+HTTPPARA="URL","https://ikfu.azurewebsites.net/api/GetUtcTime/"
    OK
    AT+HTTPACTION=0
    AT+HTTPACTION=0
    OK
    
    +HTTPACTION: 0,715,0HTTP get failed ! error code = 715
    -------------------------------------
    Request URL : https://www.ellerbach.net/DateHeure/
    AT+HTTPPARA="URL","https://www.ellerbach.net/DateHeure/"
    
    AT+HTTPPARA="URL","https://www.ellerbach.net/DateHeure/"
    OK
    AT+HTTPACTION=0
    AT+HTTPACTION=0
    OK
    
    +HTTPACTION: 0,200,20
    AT+HTTPHEAD
    
    AT+HTTPHEAD
    +HTTPHEAD: 303
    HTTP/1.1 200 OK
    Cache-Control: private
    Content-Type: text/html; charset=utf-8
    Server: Microsoft-IIS/8.5
    Set-Cookie: ASP.NET_SessionId=h2n1o2y3bt0rgnvf1wbkoxhf; path=/; HttpOnly; SameSite=Lax
    X-AspNet-Version: 4.0.30319
    X-Powered-By: ASP.NET
    Date: Sat, 05 Apr 2025 17:36:26 GMT
    Content-Length: 20
    OK
    HTTP Header : HTTP/1.1 200 OK
    Cache-Control: private
    Content-Type: text/html; charset=utf-8
    Server: Microsoft-IIS/8.5
    Set-Cookie: ASP.NET_SessionId=h2n1o2y3bt0rgnvf1wbkoxhf; path=/; HttpOnly; SameSite=Lax
    X-AspNet-Version: 4.0.30319
    X-Powered-By: ASP.NET
    Date: Sat, 05 Apr 2025 17:36:26 GMT
    Content-Length: 20
    AT+HTTPREAD?
    AT+HTTPREAD?
    +HTTPREAD: LEN,20
    AT+HTTPREAD=0,20
    
    
    OK
    AT+HTTPREAD=0,20
    OK
    
    +HTTPREAD: 20
    2025-04-05 19:36:26Z
    +HTTPREAD: 0HTTP body : 2025-04-05 19:36:26Z

I tried another SIM card of another network. Same error..

I updated the Firmware of the modem.. Same error..

I checked the validity of the certificate and it's OK:
 - DigiCert Global Root G2 is valid until 15-01-2038, 13:00:00 CET
 - Microsoft Azure RSA TLS Issuing CA 07 is valid until 26-08-2026, 01:59:59 CEST
 - *.azurewebsites.net is valid until 06-09-2025, 04:12:08 CEST

I created a simple ASP.Net Core Web API and connect to that. Same error.

My guess is that something changed in the Azure networking around 28-03-2025 and that caused the microcontroller to loose the ability to connect to my function app. But that seems very hard to believe..

I don't really understand what the error 715 "handshake failed" means.

Can you think of something?

lewisxhe, 2025-04-07:
715 There are many reasons. Only SIMCOM can analyze the log to find out what the reason is. The document only mentions that this is a handshake failure, and there is no more solution information.

You need to capture all the steps from the modem power on to connecting to Amazon, and then use https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/a7670x_capture_log.md Log Capture to send all the logs to me, and then I will send them to SIMCOM for analysis

rikkreeftenberg, 2025-04-07:
@karrisaiteja No I haven't saved certificates into the module. And I wouldn't know how to do that. I hoped that the root certificates would have been updated by the updated Firmware.
Honestly I don't think contacting the Azure team would help me. I will focus on getting more detailed logs from the modem, like @lewisxhe suggested.

rikkreeftenberg, 2025-04-07:
Capturing the logs looks complicated. Soldering a USB cable etc.. 
@lewisxhe  , do you have a A7670E?
If so, could you run the Arduino HttpsBuiltlnGet example program (https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsBuiltlnGet)?
And try GET-ing the https://ikfu.azurewebsites.net/api/GetUtcTime/  ?

I'm sure you run into the same problem.

lewisxhe, 2025-04-08:
I use the same version of firmware, but I use A7670G. The only difference is the supported area. The firmware is the same. No problems have been found so far. The following is my log

```bash
Online registration successful

Registration Status:1
Inquiring UE system information:LTE,Online,460-11,0x775C,117004605,399,EUTRAN-BAND3,1506,3,3,17,25,24,0
Network IP:10.2.209.91
Request URL : https://httpbin.org/get
HTTP Header : HTTP/1.1 200 OK
Date: Tue, 08 Apr 2025 02:45:49 GMT
Content-Type: application/json
Content-Length: 256
Connection: keep-alive
Server: gunicorn/19.9.0
Access-Control-Allow-Origin: *
Access-Control-Allow-Credentials: true
HTTP body : {
  "args": {}, 
  "headers": {
    "Accept": "*/*", 
    "Cache-Control": "no-cache", 
    "Host": "httpbin.org", 
    "X-Amzn-Trace-Id": "Root=1-67f48ddd-6e8298f138e0ebd77499387a"
  }, 
  "origin": "113.84.192.104", 
  "url": "https://httpbin.org/get"
}

-------------------------------------
Request URL : https://vsh.pp.ua/TinyGSM/logo.txt
HTTP Header : HTTP/1.1 200 OK
Server: nginx/1.22.1
Date: Tue, 08 Apr 2025 02:36:19 GMT
Content-Type: text/plain; charset=UTF-8
Content-Length: 121
Connection: keep-alive
X-Powered-By: Express
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: X-Requested-With
Cross-Origin-Resource-Policy: cross-origin
Accept-Ranges: bytes
Cache-Control: public, max-age=86400
Last-Modified: Wed, 27 Sep 2017 09:03:12 GMT
ETag: W/"79-15ec2936080"
HTTP body : 
  _____            _____  _____  _____
    |  | |\ | \_/ |  ___ |_____ |  |  |
    |  | | \|  |  |_____| _____||  |  |

-------------------------------------
Request URL : https://ipapi.co/timezone
HTTP Header : HTTP/1.1 200 OK
Date: Tue, 08 Apr 2025 02:46:08 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 13
Connection: keep-alive
Allow: OPTIONS, GET, OPTIONS, POST, HEAD
X-Frame-Options: DENY
Vary: Host, origin
X-Content-Type-Options: nosniff
Referrer-Policy: same-origin
Cross-Origin-Opener-Policy: same-origin
cf-cache-status: DYNAMIC
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=MKuG3aU%2BV1ogLEEPELaarwvnc2H%2FBXLRHs6xsG8bb2W0e3%2F3lB96py0bTy0DNcsN8wWnCmjBIqbBX%2FJy9Rb57YVCIW%2BaCoH1Gq3lj3ToC1iML5b9c%2Bs60hgy"}],"group":"cf-nel","max_age":604800}
NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Server: cloudflare
CF-RAY: 92ce6ebd9cdb1c9d-FRA
server-timing: cfL4;desc="?proto=TCP&rtt=278568&min_rtt=245089&rtt_var=102506&sent=9&recv=9&lost=0&retrans=0&sent_bytes=2825&recv_bytes=466&delivery_rate=9760&cwnd=34&unsent_bytes=0&cid=712e4fb78a642a61&ts=3275&x=0"
HTTP body : Asia/Shanghai
-------------------------------------
Request URL : http://ip-api.com/json/23.158.104.183
HTTP Header : HTTP/1.1 200 OK
Date: Tue, 08 Apr 2025 02:46:12 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 315
Access-Control-Allow-Origin: *
X-Ttl: 60
X-Rl: 44
HTTP body : {"status":"success","country":"Hong Kong","countryCode":"HK","region":"","regionName":"Kowloon","city":"Hong Kong","zip":"999077","lat":22.3193,"lon":114.169,"timezone":"Asia/Hong_Kong","isp":"Vertex Connectivity LLC","org":"Vertex Connectivity LLC","as":"AS197547 Vertex Connectivity LLC","query":"23.158.104.183"}
-------------------------------------

» AT+SIMCOMATI

« AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: xxxxxxxxxxxxxxxxxxx

OK

```

lewisxhe, 2025-04-08:
I tested the URL you provided and indeed got an error value of 715. I will capture the log and provide it to simcom
```bash
Request URL : https://ipapi.co/timezone
HTTP Header : HTTP/1.1 200 OK
Date: Tue, 08 Apr 2025 02:52:52 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 13
Connection: keep-alive
Allow: HEAD, GET, POST, OPTIONS, OPTIONS
X-Frame-Options: DENY
Vary: Host, origin
X-Content-Type-Options: nosniff
Referrer-Policy: same-origin
Cross-Origin-Opener-Policy: same-origin
cf-cache-status: DYNAMIC
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v4?s=GKZuIk0B44GDZymY1hP8oPJgsbDW7vU%2B9P12g7ULxz8Yv1zWEi48xcAyWgi7Cn%2BiuKBmB6XotQ3qiDvRoaXsruZ0%2Bsyq6BA0gF0ET1OJoFC2jR%2Fdtx2YtJaA"}],"group":"cf-nel","max_age":604800}
NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Server: cloudflare
CF-RAY: 92ce7897fc5937de-FRA
server-timing: cfL4;desc="?proto=TCP&rtt=254068&min_rtt=239742&rtt_var=64641&sent=7&recv=9&lost=0&retrans=0&sent_bytes=2827&recv_bytes=466&delivery_rate=14375&cwnd=39&unsent_bytes=0&cid=3db002122f3eb5be&ts=2999&x=0"
HTTP body : Asia/Shanghai
-------------------------------------
Request URL : http://ip-api.com/json/23.158.104.183
HTTP Header : HTTP/1.1 200 OK
Date: Tue, 08 Apr 2025 02:53:01 GMT
Content-Type: application/json; charset=utf-8
Content-Length: 315
Access-Control-Allow-Origin: *
X-Ttl: 60
X-Rl: 44
HTTP body : {"status":"success","country":"Hong Kong","countryCode":"HK","region":"","regionName":"Kowloon","city":"Hong Kong","zip":"999077","lat":22.3193,"lon":114.169,"timezone":"Asia/Hong_Kong","isp":"Vertex Connectivity LLC","org":"Vertex Connectivity LLC","as":"AS197547 Vertex Connectivity LLC","query":"23.158.104.183"}
-------------------------------------
Request URL : https://ikfu.azurewebsites.net/api/GetUtcTime
HTTP get failed ! error code = 715
Request URL : https://ikfu.azurewebsites.net/api/GetUtcTime
HTTP get failed ! error code = 715
Request URL : https://ikfu.azurewebsites.net/api/GetUtcTime
HTTP get failed ! error code = 715
-------------------------------------
```

ahoughton5, 2025-04-08:
I'm having exactly the same issue with a simcom module sending data to Azure.

This was working up until 27/03 approx 10pm GMT and suddenly started returning Http 715.

lewisxhe, 2025-04-10:
The problem has been fixed. You need to add `AT+CSSLCFG="sslversion",0,3` to configure the protocol to TLS1.2 to access AZURE, because AZURE no longer supports automatic negotiation of the SSL protocol.

```bash
Request URL : https://ikfu.azurewebsites.net/api/GetUtcTime
AT+CSSLCFG="sslversion",0,3

AT+CSSLCFG="sslversion",0,3

OK
AT+HTTPPARA="URL","https://ikfu.azurewebsites.net/api/GetUtcTime"
AT+HTTPPARA="URL","https://ikfu.azurewebsites.net/api/GetUtcTime"

OK
AT+HTTPACTION=0
AT+HTTPACTION=0

OK

+HTTPACTION: 0,200,30
AT+HTTPHEAD

AT+HTTPHEAD

+HTTPHEAD: 144
HTTP/1.1 200 OK
Content-Type: application/json; charset=utf-8
Date: Thu, 10 Apr 2025 02:54:56 GMT
Server: Kestrel
Transfer-Encoding: chunked
OK
HTTP Header : HTTP/1.1 200 OK
Content-Type: application/json; charset=utf-8
Date: Thu, 10 Apr 2025 02:54:56 GMT
Server: Kestrel
Transfer-Encoding: chunked
AT+HTTPREAD?
AT+HTTPREAD?

+HTTPREAD: LEN,30
AT+HTTPREAD=0,30

OK
AT+HTTPREAD=0,30

OK

+HTTPREAD: 30
"2025-04-10T02:54:56.0753985Z"
+HTTPREAD: 0HTTP body : "2025-04-10T02:54:56.0753985Z"
-------------------------------------
```

ahoughton5, 2025-04-10:
Have Azure confirmed this?

we figured out forcing TLS 1.2 worked just the other day, this isnt really a solution for us. We have thousands of devices out in the field which we cant connect to.  There are various Tickets open with Azure.

karrisaiteja, 2025-04-13:
@lewisxhe , I have another issue. I configured the Simcom A7608E-H module to use TLS 1.2 with the command AT+CSSLCFG="sslversion",0,3 to access Azure. This works with a regular SIM, but not with the IoT SIM, which gives error 715. The IoT service provider uses TLS 1.3, which I enabled, but it still doesn't work. Could the issue be that Azure accepts only TLS 1.2, while the IoT service provider uses TLS 1.3? Both the regular SIM and IoT SIM are provided by Airtel.

rikkreeftenberg, 2025-05-15:
The problem was that Azure restricted access to their  Web API's. So you should use: AT+CSSLCFG="sslversion",0,3

I'll close this issue.

closed 2025-05-15 by rikkreeftenberg

ahoughton5, 2025-05-15:
The Microsoft Azure Team has investigated your issue where TLS handshakes to your app were failing.
Upon investigation, engineers determined this to be caused by a regression in TLS handshake logic.
We recently made a change to our TLS handshake logic to reject requests which do not meet the MinTlsVersion requirement earlier. We did this to improve telemetry on our end and save some CPU cycles for extra processing of the handshake. However, in doing this we inadvertently exposed a bug in the framework we use to parse TLS handshakes related to SSLv3.0 Unified Client Hello.
We reverted the change in question and added extra regression testing to ensure we do not regress again. We also have an Issue opened against the .NET team: [Parse more of the ClientHello when using SSL3 in the record layer · Issue #2821 · dotnet/yarp](https://github.com/dotnet/yarp/issues/2821)
In order to protect your app in the future, we recommend checking out [this guide](https://azure.github.io/AppService/2020/05/15/Robust-Apps-for-the-cloud.html) for ideas about how to be as resilient as possible.
We apologize for any inconvenience.