Xinyuan-LilyGO/LilyGo-Modem-Series #244: Traccar Example - GPS randomly loses fix after modem sleep

Masssssy, 2025-04-06:
When running the Traccar example code everything seems to work fine when USB is connected. 

When running off battery the GPS fix is lost after 10-60 minutes. The GPS fix never comes back. If the device isreset, GPS fix will again be aquired and lost within 10-60 minutes. 

Since the device A7670E does not seem to enter sleep when USB is connected my guess is that something goes wrong in the sleep/wake process and GPS becomes inactive somehow.

lewisxhe, 2025-04-07:
Thanks for the report, I only connected the USB power supply to test, and did not use the battery test, I will test it later.

lewisxhe, 2025-04-07:
[image]
I used SIM7670G for the test. I just used A7670SA for the test. Apart from finding that the current was abnormal (it seemed that it did not enter sleep), it was able to upload the current GPS coordinates at the preset 20-second interval.

closed 2025-05-23 by github-actions[bot]