Xinyuan-LilyGO/LilyGo-Modem-Series #245: Problem to MQTT AWS connection

cucilo, 2025-04-08:
HI, I use LILYGO T-CALL A7670 v1.1 and I have problem to MQTT AWS connection. When I compile, I have this warning:

```
In file included from /Users/vladocucilo/Desktop/LilyGO-T-A76XX-main 2/examples/MqttsBuiltlnAWS/MqttsBuiltlnAWS.ino:22:
examples/MqttsBuiltlnAWS/utilities.h:25: warning: "LILYGO_T_CALL_A7670_V1_1" redefined
   #define LILYGO_T_CALL_A7670_V1_1
 
<command-line>: note: this is the location of the previous definition
```

because I uncommented #define LILYGO_T_CALL_A7670_V1_1 in file utilities.h

And my log is:

```
AT
AT

*ATREADY: 1
AT
AT
OK

AT+CPIN?
AT+CPIN?
+CME ERROR: 13

*ISIMAID: "A0000000871004F387FF0189B00020"

+CPIN: READY
SIM card online
AT+CNMP=2

+CGEV: EPS PDN ACT 1
AT+CNMP=2
OK
AT+CNMP?
AT+CNMP?
+CNMP: 2

OK
Current network mode : AUTO
Set network apn : mtelfrend
AT+CGDCONT=1,"IP","mtelfrend"
AT+CGDCONT=1,"IP","mtelfrend"
OK
Wait for the modem to register with the network.AT+CEREG?
AT+CEREG?
+CEREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?

SMS DONE
AT+CPSI?
+CPSI: LTE,Online,218-05,0x1FE,309262,68,EUTRAN-BAND3,1733,2,0,12,60,56,12

OK
Inquiring UE system information:LTE,Online,218-05,0x1FE,309262,68,EUTRAN-BAND3,1733,2,0,12,60,56,12
AT+NETOPEN
AT+NETOPEN
OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR
+IPADDR: 10.219.35.xxx

OK
Network IP:10.219.35.xxx
AT+CMQTTSTART
AT+CMQTTSTART
OK

+CMQTTSTART: 0

PB DONE
Connecting to xxxx.iot.eu-west-3.amazonaws.comAT+CCERTDOWN="ca_cert.pem",1201
AT+CCERTDOWN="ca_cert.pem",1201
>
-----BEGIN ROOT CERTIFICATE-----
xxxxxxx
//-----END ROOT CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
AT+CSSLCFG="cacert",0,"ca_cert.pem"
OK
AT+CCERTDOWN="cert.pem",1225
AT+CCERTDOWN="cert.pem",1225
>
-----BEGIN CERTIFICATE-----
xxxxxxx
-----END CERTIFICATE-----

OK
AT+CSSLCFG="clientcert",0,"cert.pem"
AT+CSSLCFG="clientcert",0,"cert.pem"
OK
AT+CCERTDOWN="key_cert.pem",1680
AT+CCERTDOWN="key_cert.pem",1680
>
-----BEGIN RSA PRIVATE KEY-----
xxxxxxx
-----END RSA PRIVATE KEY-----

OK
AT+CSSLCFG="clientkey",0,"key_cert.pem"
AT+CSSLCFG="clientkey",0,"key_cert.pem"
OK
AT+CSSLCFG="sslversion",0,4
AT+CSSLCFG="sslversion",0,4
OK
AT+CMQTTSSLCFG=0,0
AT+CMQTTSSLCFG=0,0
OK
AT+CSSLCFG="authmode",0,2
AT+CSSLCFG="authmode",0,2
OK
AT+CMQTTREL=0
AT+CMQTTREL=0
+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
AT+CMQTTACCQ=0,"A76XX",1
OK
AT+CMQTTCFG="version",0,4
AT+CMQTTCFG="version",0,4
OK
AT+CMQTTCONNECT=0,"tcp://xxx.iot.eu-west-3.amazonaws.com:8883",60,1
AT+CMQTTCONNECT=0,"tcp://xxx.iot.eu-west-3.amazonaws.com:8883",60,1
+CMQTTCONNECT: 0,34

ERROR
Failed!
```

cucilo, 2025-04-08:
```
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: A7670E-FASE
Revision: A110B01A7670M7_F
A7670M7_B01V01_240105
```

cucilo, 2025-04-10:
Can you explan me how to update firmware on this card. On video, you have 2 usb port. On my card, I have only one (LILYGO-T-CALL_A7670 v1.1). When I start A76XX_A79XX_MADL v1.13, hold boot and plug in usb to PC or card, I cant to see anything in Connect device, list is empty....

lewisxhe, 2025-04-11:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/180

cucilo, 2025-04-30:
When I connect for upgrade, into device manager, I saw message Device description request failed and when I try to install driver, I cant do anything. When I run A76xx...MADL v1.13, Device is recognise, but progress is 0%, and status is downloading data....

[image]
[image]

lewisxhe, 2025-05-05:
Before T-Call is powered on, press and hold the 7670_BOOT button. Before the modem starts, keep pressing the 7670_BOOT button until the USB is recognized and then release it. The modem will then enter the download update mode.

cucilo, 2025-05-06:
Solved. Usb cable must be directly connected on board (remove usb board from image).

cucilo, 2025-05-06:
I upgraded firmware. But when I start MQTT AWS connection, I received same log as previous (first message here).....

[image]

lewisxhe, 2025-05-07:
```bash
« AT+CPSI?

+CGEV: ME PDN ACT 8,1
AT+CPSI?

+CPSI: LTE,Online,460-11,0x775C,27159101,62,EUTRAN-BAND8,3740,3,0,8,49,53,-3

OK
Inquiring UE system information:LTE,Online,460-11,0x775C,27159101,62,EUTRAN-BAND8,3740,3,0,8,49,53,-3
AT+NETOPEN
AT+NETOPEN

OK

+NETOPEN: 0AT+IPADDR

SMS DONE
AT+IPADDR

+IPADDR: 10.21.157.148

OK
Network IP:10.21.157.148
AT+SIMCOMATI
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B07A7670M7_F
A7670M7_B07V01_240927
QCN:
IMEI: 862771076231466
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
AT+CMQTTSTART
AT+CMQTTSTART

OK

+CMQTTSTART: 0
Connecting to xxxxxx.iot.ap-southeast-2.amazonaws.com

AT+CCERTDOWN="ca_cert.pem",1189
AT+CCERTDOWN="ca_cert.pem",1189

>
-----BEGIN CERTIFICATE-----
xxxxxxxxxxxxx
-----END CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
AT+CSSLCFG="cacert",0,"ca_cert.pem"

OK
AT+CCERTDOWN="cert.pem",1221
AT+CCERTDOWN="cert.pem",1221

>
-----BEGIN CERTIFICATE-----
xxxxxxxxxxxxx
-----END CERTIFICATE-----

OK
AT+CSSLCFG="clientcert",0,"cert.pem"
AT+CSSLCFG="clientcert",0,"cert.pem"

OK
AT+CCERTDOWN="key_cert.pem",1676
AT+CCERTDOWN="key_cert.pem",1676

>
-----BEGIN RSA PRIVATE KEY-----
xxxxxxxxxxxxx
-----END RSA PRIVATE KEY-----

OK
AT+CSSLCFG="clientkey",0,"key_cert.pem"
AT+CSSLCFG="clientkey",0,"key_cert.pem"

OK
AT+CSSLCFG="sslversion",0,4
AT+CSSLCFG="sslversion",0,4

OK
AT+CMQTTSSLCFG=0,0
AT+CMQTTSSLCFG=0,0

OK
AT+CSSLCFG="authmode",0,2
AT+CSSLCFG="authmode",0,2

OK
AT+CMQTTREL=0
AT+CMQTTREL=0

+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
AT+CMQTTACCQ=0,"A76XX",1

OK
AT+CMQTTCFG="version",0,4
AT+CMQTTCFG="version",0,4

OK
AT+CMQTTCONNECT=0,"tcp://xxxxxx.iot.ap-southeast-2.amazonaws.com:8883",60,1
AT+CMQTTCONNECT=0,"tcp://xxxxxx.iot.ap-southeast-2.amazonaws.com:8883",60,1

OK

+CMQTTCONNECT: 0,0successfully.
AT+CMQTTDISC?

AT+CMQTTDISC?

+CMQTTDISC: 0,0
+CMQTTDISC: 1,1

OK
MQTT has connected!
AT+CMQTTSUB=0,21,0,0
AT+CMQTTSUB=0,21,0,0

>GsmMqttTest/subscribe

OK

+CMQTTSUB: 0,0
AT+CMQTTTOPIC=0,19
AT+CMQTTTOPIC=0,19

>GsmMqttTest/publish

OK
AT+CMQTTPAYLOAD=0,10
AT+CMQTTPAYLOAD=0,10

>RunTime:29

OK
AT+CMQTTPUB=0,0,60
AT+CMQTTPUB=0,0,60

OK

+CMQTTPUB: 0,0

+CMQTTRXSTART: 0,21,3
+CMQTTRXTOPIC: 0,21
GsmMqttTest/subscribe
+CMQTTRXPAYLOAD: 0,3
222
+CMQTTRXEND: 0
======mqtt_callback======
Topic:GsmMqttTest/subscribe
Payload:
32,32,32,
=========================

```
The same version, the test is fine, please check whether your certificate is correct.

cucilo, 2025-05-07:
Do you have some video how to setup Iot AWS to work with this card. Process must me very easy, but for me dont work. I receive same error every time.

cucilo, 2025-05-08:
solved.