Xinyuan-LilyGO/LilyGo-Modem-Series #247: A7670E failing SendSMS example

MrSuicideParrot, 2025-04-10:
I have been trying to run the SendSMS example without sucess. I have tried multiple SIM cards and the error persists. I even defined the APN,
If I run the code without changing anything, it fails like this:
```
Start modem...
AT

*ATREADY: 1

+CPIN: READY
AT
OK
Wait SMS Done.
Can't wait from sms register ....
```
I commented the part that was failing the (`modem.waitResponse(100000UL, "SMS DONE")` ).  And even there it continue to fail when it was trying to send the sms.

```
Start modem...
AT

*ATREADY: 1

+CPIN: READY
AT
OK
Wait for the modem to register with the network.AT+CEREG?
AT+CEREG?
+CEREG: 0,11

OK
AT+CGREG?
AT+CGREG?
+CGREG: 0,11

OK
Registration Status:11

Registration Status:11
                      Init success, start to send message to  +351919422367
AT+GSN
AT+GSN
862771076566085

OK
AT+CMGF=1
AT+CMGF=1
ERROR
AT+CSCS="GSM"
AT+CSCS="GSM"
OK
AT+CMGS="+35191XXXXXX"
AT+CMGS="+35191XXXXXX"
ERROR
Send sms message fail
AT+CREG?
+CREG: 0,11

OK
```

It looks like it is registering successfully, but when it does  `AT+CMGF=1` it answers with an error. 

The board in question is a T-A7670E 4G without GPS.

```
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: A7670E-LNXY-UBL
Revision: 23110B01A7670M4A_CUS_UBIA_91530228
A7670M4_B01V01_230724
IMEI: 862771076566085
```
Any suggestions about what is happening?

lewisxhe, 2025-04-11:
Cannot wait until SMS DONE means that SMS service cannot be registered. Are you sure that the SMS card you are using supports SMS service? Have you changed the SIM card to test it?

MrSuicideParrot, 2025-04-11:
Ah ok I didn't know that. I have testes with two different SIM cards, with the same results. And I have tested right know with my mobile phone, and I was able to send SMS's with them.

lewisxhe, 2025-04-14:
I noticed your version is `A7670E-LNXY-UBL`, this version does not support voice and SMS functions.

MrSuicideParrot, 2025-04-14:
Thank you very much!