Xinyuan-LilyGO/LilyGo-Modem-Series #248: SD card doesn't work when the boad is powered by battery

davidarbelaez, 2025-04-10:
Hello

I used the SDcard.ino example, and it worked well when the board was powered by the USB cable, the data was saved on the micro SD card with no problem. Then I tryed powering the board with battery and it does work, and does everything I programmed, except to save in the micro SD card, it doesn't save anything when it is powered by battery. 

How can I solve that, help please, I really need this board work with baterries and also that save data on a micro SD card.

Thank you, I will be attentive to your answer.

lewisxhe, 2025-04-11:
Add a power-on operation at the beginning of setup. You should be using the T-A76XX-ESP32 version.
```bash
void setup()
{
    Serial.begin(115200);

#ifdef BOARD_POWERON_PIN
    // T-A7670X V1.x SD card power supply and Modem are controlled by IO12. When using battery, IO12 must be set to HIGH
    pinMode(BOARD_POWERON_PIN, OUTPUT);
    digitalWrite(BOARD_POWERON_PIN, HIGH);
#endif

    SPI.begin(BOARD_SCK_PIN, BOARD_MISO_PIN, BOARD_MOSI_PIN);
    if (!SD.begin(BOARD_SD_CS_PIN)) {
        Serial.println("Card Mount Failed");
        return;
    }
}

```

davidarbelaez, 2025-04-11:
Thank you very much @lewisxhe, it solved the problem!

closed 2025-05-13 by lewisxhe