Xinyuan-LilyGO/LilyGo-Modem-Series #249: Flow meter T-SIM7670-S3

MatiasRaya, 2025-04-11:
Hi, I wanted to know if the t-sim7670-s3 can connect directly to
https://www.duaitek.com.ar/caudalimetro-sensor-flujo-03-a-6lmin-yf-s401-arduino-gtia/p/MLA27393316?pdp_filters=seller_id%3A39384567#polycard_cl ient=search-nordic-mshops&position=1&search_layout=stack&type=item&tracking_id=83a19758-eb3f-4073-a41e-dab3e5f6a458&wid=MLA1528060462&sid=search

The question arises because the sensor is powered by 5V, and I don't know the output voltage. If it's 5V, should I use a divider? resistive or does the board support a 5V input?

lewisxhe, 2025-04-14:
t-sim7670-s3 supports 5V input, only 3.3V output, no 5V output

MatiasRaya, 2025-04-14:
Would it be possible for you to share a schematic of how I should connect it? I'm afraid of ruining my board with the 5V input to the flowmeter.

lewisxhe, 2025-04-15:
[image]

MatiasRaya, 2025-04-15:
I know it can be powered with 5V. I was referring to GPIOs, such as GPIO 40/41. That one works with a 3.3V input from a sensor, right? I'm attaching an image as an example.

[image]

lewisxhe, 2025-04-16:
As long as there is no assigned GPIO, it can be used freely.

closed 2025-04-17 by MatiasRaya