Xinyuan-LilyGO/LilyGo-Modem-Series #254: Lilygo A7670E can't sms register

alexlopez2612, 2025-04-16:
I want to send / receive sms and I get the same problem as commented on the post:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/247
On that post, somebody says that:
"I noticed your version is A7670E-LNXY-UBL, this version does not support voice and SMS functions."
I got the same version, so, I understand I have the same problem.

This is my version:
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-LNXY-UBL
Revision: 23110B01A7670M4A_CUS_UBIA_91530228
A7670M4_B01V01_230724
IMEI: 862771076563371

It is possible to update the firmware to be able to send/receive SMS?
Where can I get the correct version?
Thank for your reply
Regards

lewisxhe, 2025-04-17:
The ability to send text messages and make calls cannot be obtained through software upgrades.
You must purchase the version with voice and SMS functions to have these features

alexlopez2612, 2025-04-17:
Thanks @lewisxhe 
I bougth this from the lilygo store on Aliexpress, and now Im checking and I don't see that this version is limited on this features, so I thougt that this board would work with sms.
So, how I can know which version is the one I need? On lilygo oficial web site I cant found this info either.

Thank you!
REgards

lewisxhe, 2025-04-18:
https://lilygo.cc/products/t-sim-a7670e?_pos=1&_sid=f70e3263b&_ss=r&variant=42534734495925
The lilygo store does not have a separate data version, all versions can be used for voice calls and text messages

closed 2025-06-01 by github-actions[bot]