Xinyuan-LilyGO/LilyGo-Modem-Series #255: LilyGO T-Call A7670E GPS is not responding

AVEgraff, 2025-04-17:
Hello! I purchased the above module. I downloaded various firmware several times to understand how the module works and to check its functionality. I found several issues.
The GSM module works properly, sends messages and responds to the port monitor.
Complaints about the GPS operation. The GPS is listed as faulty in the port monitor. I measured the voltage at the GPS antenna output, everything is fine, about 3V. The module itself receives a voltage of 4.2V from the USB-typeC power supply. I am attaching the AT command response logs.

=== ТЕСТ ESP32 в LilyGo T-Call A7670E ===
ID Chip: B21C0610
Frequency CPU: 240 MHz
Ядер: 2
Свободная RAM: 276388 байт
Flash: 4 MB
Test GPIO...
Напряжение VBAT: 0.00V
MAC WiFi: 10:06:1C:B2:BE:70
UART1 не отвечает (SIMCOM выключен?)

=== ТЕСТ ЗАВЕРШЕН ===
Work.Heap: 228048 байт

AT+CGMR

+CGMR: A011B04A7670M7_F

Напряжение VBAT: 0.00V
MAC WiFi: 10:36:1D:B2:RE:70

AT+SIMCOMATI
<< Manufacturer: INCORPORATED
<< Model: A7670E-FASE
<< Revision: A011B04A7670M7_F
<< A7670M7_B04V02_220927
<< QCN:
<< IMEI: 862771072427704
<< MEID:
<< +GCAP: +CGSM,+FCLASS,+DS
<< DeviceInfo:
<< OK
<< +CSQ: 25,99
<< OK

AT+CGPS?    AT+CGPS=1,1    AT+CGPSINFO   AT+CGPSSTATUS?   
The module responds to all commands regarding GPS --- ERROR

Please advise what additional information is required on the operation of the module to resolve the issue with GPS

lewisxhe, 2025-04-18:
You don't have GPS turned on, so sending GPS commands will of course return errors, have you used the examples provided to troubleshoot? Or have you looked at the AT command manual?

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/137

AVEgraff, 2025-04-18:
Hello! I used some examples, but there is no result. From the example, I used a piece of code, as I understand it, to wake up the module. 
This piece.
    // Turn on modem
    pinMode(BOARD_PWRKEY_PIN, OUTPUT);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, HIGH);
    delay(1000);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);

Below is the response to the AT commands.

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A7670M7_V1.11.1
IMEI: 862771072425591
+GCAP: +CGSM,+FCLASS,+DS

OK
>> +CGDCONT=1
>> +CGDCONT=1
>> AT+CGNSPWR?

ERROR
>> AT+CGNSPWR=1

ERROR
>> AT+CGNSSTART

ERROR
>> AT+CGNSMOD=<1>

ERROR
>> AT+CGNSMOD=<2>

ERROR
>> AT+CGNSAUTOSTART=<1>

ERROR
>> AT+CGNSTST=1

ERROR
>> AT+CGNSRST

ERROR

Additionally, I will now review the examples given again.

AVEgraff, 2025-04-18:
The revised example partially helped, periodically the port monitor receives information that it was possible to turn on the modem, but this is periodic. Not constantly and steadily.

Modem is ready
Enabling GPS...
GPS enabled successfully
No GPS fix available
Waiting for modem...

Modem is ready
Enabling GPS...
Failed to enable GPS!

lewisxhe, 2025-04-18:
Are you T-CALL or T-PCIE? ?

lewisxhe, 2025-04-18:
Please run the sample code to see the results
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/GPS_BuiltIn/GPS_BuiltIn.ino

AVEgraff, 2025-04-18:
> Are you T-CALL or T-PCIE? ?

Lillyho T-Call A7670E board

I had difficulty compiling the example and eventually reworked it without changing its essence. I am attaching a revised example. As well as responses to his commands.

#define TINY_GSM_MODEM_A7670    // Определение модели модема
#define TINY_GSM_RX_BUFFER 1024 // Размер буфера приёма

// Распиновка для LilyGO T-Call A7670 V1.1
#define MODEM_TX_PIN     27
#define MODEM_RX_PIN     26
#define BOARD_PWRKEY_PIN 4
#define MODEM_RESET_PIN  5
#define BOARD_LED_PIN    13

// ========== БИБЛИОТЕКИ ==========
#include <TinyGsmClient.h>

// Используем встроенный Serial1 для модема
#define SerialAT Serial1
TinyGsm modem(SerialAT);

void setup() {
  // Инициализация последовательного порта для отладки
  Serial.begin(115200);
  delay(1000);
  
  // Настройка пинов
  pinMode(BOARD_LED_PIN, OUTPUT);
  digitalWrite(BOARD_LED_PIN, HIGH);

  // Запуск модема
  Serial.println("Initializing modem...");
  powerOnModem();
  
  // Инициализация UART для модема
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);
  delay(3000);

  // Проверка связи с модемом
  if (!waitForModem()) {
    Serial.println("Failed to initialize modem!");
    while(1);
  }

  // Включение GPS
  Serial.println("Enabling GPS...");
  if (!modem.enableGPS()) {
    Serial.println("Failed to enable GPS!");
    while(1);
  }
  Serial.println("GPS enabled successfully");
}

void loop() {
  // Переменные для хранения GPS данных
  float lat, lon, speed, alt, accuracy;
  int vsat, usat, year, month, day, hour, min, sec;
  uint8_t fixMode;

  // Получение GPS данных
  if (modem.getGPS(&fixMode, &lat, &lon, &speed, &alt, &vsat, &usat, &accuracy,
                  &year, &month, &day, &hour, &min, &sec)) {
    printGPSData(fixMode, lat, lon, speed, alt, vsat, usat, accuracy,
                year, month, day, hour, min, sec);
                
    // Вывод сырых NMEA данных
    String rawData = modem.getGPSraw();
    Serial.print("Raw NMEA data: ");
    Serial.println(rawData);
  } else {
    Serial.println("No GPS fix available");
  }

  delay(5000); // Задержка между опросами
}

// ========== ВСПОМОГАТЕЛЬНЫЕ ФУНКЦИИ ==========

void powerOnModem() {
  pinMode(BOARD_PWRKEY_PIN, OUTPUT);
  digitalWrite(BOARD_PWRKEY_PIN, LOW);
  delay(100);
  digitalWrite(BOARD_PWRKEY_PIN, HIGH);
  delay(1000);
  digitalWrite(BOARD_PWRKEY_PIN, LOW);
}

bool waitForModem() {
  Serial.println("Waiting for modem...");
  int retry = 0;
  while (!modem.testAT(1000)) {
    Serial.print(".");
    if (retry++ > 10) {
      // Перезагрузка модема при проблемах
      powerOnModem();
      retry = 0;
    }
  }
  Serial.println("\nModem is ready");
  return true;
}

void printGPSData(uint8_t fix, float lat, float lon, float speed, float alt,
                 int vsat, int usat, float accuracy, int year, int month,
                 int day, int hour, int min, int sec) {
  Serial.println("\n=== GPS DATA ===");
  Serial.print("Fix: "); Serial.println(fix);
  Serial.print("Location: ");
  Serial.print(lat, 6); Serial.print(", "); Serial.println(lon, 6);
  Serial.print("Speed: "); Serial.print(speed); Serial.println(" km/h");
  Serial.print("Altitude: "); Serial.print(alt); Serial.println(" m");
  Serial.print("Satellites: "); Serial.print(usat); Serial.print("/"); Serial.println(vsat);
  Serial.print("Accuracy: "); Serial.println(accuracy);
  Serial.print("Date: ");
  Serial.printf("%02d.%02d.%04d\n", day, month, year);
  Serial.print("Time: ");
  Serial.printf("%02d:%02d:%02d\n", hour, min, sec);
}

Result:
Modem is ready
Enabling GPS...
Failed to enable GPS!

AVEgraff, 2025-04-18:
If we take only the example from the link that you provided, then I am attaching information from the monitor port.

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: A011B04A7670M7_F
A7670M7_B04V02_220927
QCN:
IMEI: 862771072427704
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK

OK

+CGNSSPWR: READY!

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.

lewisxhe, 2025-04-18:
« Start modem...
AT
AT
AT
AT
.
AT
AT
AT
AT
.
AT

*ATREADY: 1
[9892] ### Unhandled: *ATREADY: 1
AT
AT

OK

ATE0
ATE0

OK
ATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: FAETEST_V1.0.0
IMEI: 
+GCAP: +CGSM,+FCLASS,+DS

OK
[10220] ### Modem: A7670E-FASE
Model Name:A7670E-FASE
AT+SIMCOMATI

Manufacturer: INCORPORATED
Model: A7670E-FASE
Revision: xxxxxxxxx
FAETEST_B01V01_2XXXX
QCN:
IMEI: 
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:

OK
Enabling GPS/GNSS/GLONASS
AT+CGNSSPWR=1

OK

+CPIN: SIM REMOVED

+CGNSSPWR: READY!
GPS Enabled
AT+CGNSSIPR=115200

OK
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: ,,,,,,,,

OK
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
Requesting current GPS/GNSS/GLONASS location
AT+CGNSSINFO

+CGNSSINFO: 3,15,,05,00,xx.xxxxx778,N,xxxx.xxxxx9518,E,180425,072255.00,169.9,0.379,,10.40,6.39,8.20,0

OK
FixMode:3
Latitude:22.654278	Longitude:114.059952
Speed:0.38	Altitude:169.90
Visible Satellites:15	Used Satellites:0
Accuracy:6.39
Year:2025	Month:4	Day:18
Hour:7	Minute:22	Second:55
Retrieving GPS/GNSS/GLONASS location again as a string
AT+CGNSSINFO

+CGNSSINFO: 3,15,,06,00xxxx.xxxxxx37,N,xxxx.xxxxxx3,E,180425,072258.00,199.9,8.588,197.77,10.39,6.39,

OK
GPS/GNSS Based Location String:3,15,,06,00,xxx.xxxxx7,N,xxx.xxxxx,E,180425,072258.00,199.9,8.588,197.77,10.39,6.39,
Disabling GPS
AT+CGNSSPWR=0

OK

AVEgraff, 2025-04-18:
Do I understand correctly that the system is in working order, and the coordinates of the satellites are simply not being received because the module is in cold mode and has not yet received the coordinates?

lewisxhe, 2025-04-18:
These boards do not have hot start function, all are cold start

AVEgraff, 2025-04-18:
The SIM card is installed, the module is located on the street, I will wait for the coordinates to be received. Thank you. I'll write you the results.

AVEgraff, 2025-04-18:
40 minutes have passed, the result has not changed. :/

AVEgraff, 2025-04-18:
A question has arisen. The board is currently in satellite signal search mode (so far), if I take and flash a sketch that sends commands to the port monitor and send commands regarding GPS (AT+CGNSPWR=1; AT+CGNSPWR?), then I also get an ERROR response to these commands. Does the GPS turn off in time? Or is there a different logic to the work? Tell me please.

lewisxhe, 2025-04-18:
It is AT+CGNSSPWR=1

Instead of AT+CGNSPWR=1, an S is missing

AVEgraff, 2025-04-18:
Thanks! I've looked at that. Study the A7600 series computer In the instruction manual

AVEgraff, 2025-04-19:
Good afternoon! I want to tell you about the results of the tests. In general, the position of the module near the window did not give any improvements, in this case, LBS location detection only saved. But when I went outside, I detected 3 satellites within 5 minutes. In urban conditions, this accuracy is comparable to LBS. Let's try other GPS antennas. Thanks to the LilyGO team for responding to the issue!

lewisxhe, 2025-04-21:
You're welcome.

closed 2025-05-13 by lewisxhe