Xinyuan-LilyGO/LilyGo-Modem-Series #256: ESP32 A7670e problem in using I2C

p-sio, 2025-04-18:
Hello! I am having problem when using MCP32017 (multiplexer module liek this: https://protosupplies.com/product/mcp23017-16-bit-i-o-expander-i2c-module/) and OLED 1.3"

when i am using the GPIO from esp32 board, it works fine, but when I'm using the pins in multiplexer i am getting these:

[image]

Now, i tried using oled + mcp in other code and it's working, although i am getting uncalled reboots and this:

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400844c4  PS      : 0x00060033  A0      : 0x80083a10  A1      : 0x3ffbfbbc  
A2      : 0x3ffb850c  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffb887c  
A6      : 0x3ffb8560  A7      : 0x3ffbfbd8  A8      : 0x00000000  A9      : 0x3ff53000  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00060023  
A14     : 0x00000000  A15     : 0x0000abab  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40086094  LEND    : 0x400860aa  LCOUNT  : 0x00000000  

Backtrace: 0x400844c1:0x3ffbfbbc |<-CORRUPTED

Does this have something to do with SLC SDA? if not, what seems to be the issue?

here's my code,
for the first problem:

#include <Adafruit_MCP23017.h>
#include <Wire.h>
#include <utilities.h>
#include <U8g2lib.h>

#define TINY_GSM_DEBUG SerialMon
#define SerialMon Serial

#define DUMP_AT_COMMANDS
#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

Adafruit_MCP23017 mcp;

#define BUTTON_A_PIN 1
#define BUTTON_B_PIN 2
#define BUTTON_C_PIN 3
#define BUTTON_D_PIN 4

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);

String messages[40]; 
int simIndexes[40];  
int currentMessageIndex = 0;

#define DEBOUNCE_DELAY 20
unsigned long lastButtonPress = 0; 

enum Mode { INBOX_MODE, SELECTED_MESSAGE };
Mode currentMode = INBOX_MODE;

unsigned long lastFetchTime = 0;
const unsigned long fetchInterval = 10000;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Wire.begin(21, 22); // Use your actual I2C pin numbers
[... 520 lines cut ...]
        typedText = ""; // Clear message buffer
    } else {
        Serial.println("No message to send.");
    }
}

void updateDisplay() {
    u8g2.clearBuffer();

    // Show typed message (trim if it's long)
    String displayText = typedText;
    if (displayText.length() > 18) {
        displayText = displayText.substring(displayText.length() - 18); // show last 18 chars
    }

    u8g2.setCursor(0, 12);
    u8g2.print("Text: ");
    u8g2.print(displayText);

    u8g2.setCursor(0, 30);
    u8g2.print("Mode: ");
    u8g2.print(isNumberMode ? "Number" : "Letter");

    u8g2.sendBuffer();
}

lewisxhe, 2025-04-19:
Are you using ESP32-A7670?
21, 22 are free and not used. From the assertion, it can be seen that the program has illegal operations that cause the crash.
You need to use [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to find the crash site

p-sio, 2025-04-19:
> Are you using ESP32-A7670? 21, 22 are free and not used. 

Thanks for replying. May I ask what is meant by 21, 22 being free and not used?

is something wrong with this part?

[image]

wire.begin(<sda pin #>, <scl pin #>);

and for EspExceptionDecoder, I will try it. Thank you!

lewisxhe, 2025-04-21:
GPIO21,22 are idle and not connected to other peripherals. You can use them freely, so this is not caused by the board.

p-sio, 2025-04-25:
May I ask if pin 21 and 22 (for SDA, SCL) have internal pull-up resistors?

lewisxhe, 2025-04-25:
There is no pull-up resistor, but generally external modules will have a pull-up resistor.

p-sio, 2025-04-25:
Aw yep, i checked and it's only sda, scl (not gpio). Thank you! Do you have a datasheet for the esp32 not just the simcon for a7670e lte? Or datasheet for the whole board esp32 A7670e?

lewisxhe, 2025-04-25:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/schematic/T-A7670X-V1.1.pdf
https://www.espressif.com.cn/en/support/download/documents/modules?keys=&field_type_tid%5B%5D=681

p-sio, 2025-05-01:
Hello! In case anyone comes across the same problem as mine, I want to share that I have solved this problem simply by reverting my esp32 version into an older one (the one found in boards manager in arduino IDE). The problem was not in the code, but the latest version of esp32 in arduinoIDE (2.3.0)

closed 2025-05-01 by p-sio