Xinyuan-LilyGO/LilyGo-Modem-Series #258: Seeking Solution for Storing and Resending Failed HTTPS Requests in IoT Application

karrisaiteja, 2025-04-22:
Hi,

I am currently using the TTGO-T-Sim7608E-H to push data to the cloud. However, sometimes network issues cause the system to fail in sending the data. The device goes to sleep for 1 hour, then wakes up to gather all sensor data and push it to the cloud. Currently, I am storing the sensor data on an SD card.

I am looking for a solution to store failed HTTPS request data in a suitable file format. When the device wakes up in the next 1-hour interval, it should check for any historical data and send it to the cloud if present. Can anyone suggest the best solution for this kind of application?

lewisxhe, 2025-04-22:
This depends on what format your server is willing to accept.

karrisaiteja, 2025-04-23:
@lewisxhe below is the format of the server accepting the data CreatedDate will log the sensor data with the historical timestamp. 

URL https://testing.azurewebsites.net/api/SensorReading/IngestData

Headers

Content-Type:  application/json

x-functions-key:  xxxxxxxxxxxxxxxxxxxxxxxxxxxx

Body

{
"UUID": "0000-0000-x010",
"CreatedDate": "2024-04-07T19:27",
"Tag": "Weather Station",
"PayLoad": "{
     \"temperature\": \"26.54\",
     \"humidity\": \"88.2\"
     }"
}

lewisxhe, 2025-04-23:
How to save the data? You can choose by yourself. You can save the string or just save the data and timestamp, and then combine them into JSON format when uploading.

closed 2025-05-13 by lewisxhe