Xinyuan-LilyGO/LilyGo-Modem-Series #259: YF-S201 flow sensor T-SIM7670G-S3

MatiasRaya, 2025-04-22:
Hi, has anyone ever connected the YF-S201 flow sensor to the board? 
The datasheet for the sensor is: https://www.dataq.com/products/accessories/flow-sensor/2000362.html?srsltid=AfmBOoqsnLBfOrR6lw97FygcBBQUqMxZ9tldcxG-uiiNhcKodqI9Ir8X

lewisxhe, 2025-04-23:
It looks like this is a flow sensor that can be driven by inputting a PWM signal. The manual says that it outputs a 5V TTL signal, so this interface should be 5V. The safety voltage of ESP32 is 3.3V. You need a PWM isolator. You can try to use ESP to output a PWM signal and provide a flow sensor through the PWM isolator.
However, you still need to confirm with the merchant whether the flow sensor can be driven by outputting a PWM signal.

[image]

MatiasRaya, 2025-04-23:
Hello, I didn't understand your answer, could I rephrase it?

lewisxhe, 2025-04-23:
1. First, you need to communicate with the merchant of this flow sensor and ask whether this sensor is driven by PWM signal and has 5V TTL interface.
2. If the reply is that the flow sensor is driven by PWM signal, then you need a 3.3V TO 5V isolator, and ESP drives the sensor through the isolator

MatiasRaya, 2025-04-23:
Talk to the manufacturer and I am told that pulses are observed at the output, ranging between 0V and 5V. A fact that I can personally check with an oscilloscope, as the flow rate increases, the period between square pulses decreases/increases. 
My doubt is how to read the 5V or 12V output on the board since I understand that it supports up to 3.3V

lewisxhe, 2025-04-23:
Ok, I misunderstood, this is the input, you need to input a voltage signal of up to 5V to the ESP. You need a resistor divider to divide the 5V into the voltage range that the ESP can receive, two 100K resistors will do, similar to the battery voltage detection.

MatiasRaya, 2025-04-23:
Thank you very much, I will test with the resistive divider. 

With respect to GPIO15 use, is it okay? or should I use another?

lewisxhe, 2025-04-23:
IO15 is free and can be used

MatiasRaya, 2025-04-23:
perfect, I will be performing the relevant tests and update the issue 

Thank you so much for your help.

MatiasRaya, 2025-04-24:
Hello, I update the status of this issue. after measuring and analyzing both with the oscilloscope and with other instruments, I saw that it was safe to connect the direct flowmeter to the plate. So far I had no problems with that GPIO and the caudalimeter measures quite accurate. 

Thank you for the help.

closed 2025-04-24 by MatiasRaya