Xinyuan-LilyGO/LilyGo-Modem-Series #260: Multithread for GPS use

MatiasRaya, 2025-04-24:
Hello, I wanted to know if you can move the positioning of the GPS to a separate thread so as not to affect the main loop.

MatiasRaya, 2025-04-24:
The solution was

void gpsTask(void *pvParameters)
{
    while (true) {
        // String numberCore = String(xPortGetCoreID());
        // LOG_INFO(classNAME, "GPS task in core: %s", numberCore.c_str());
        
        if (enableGPS) {
            if (getGPSLocation(latitude, longitude)) {
                LOG_INFO(classNAME, "GPS location: %f, %f", latitude, longitude);
            } else {
                LOG_ERROR(classNAME, "Failed to get GPS location");
            }
        }

        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

and in the setup of my main file I had to put

xTaskCreatePinnedToCore(
        gpsTask,
        "gpsTask",
        10000,
        NULL,
        1,
        &gpsTaskHandle,
        0
    );

lewisxhe, 2025-04-25:
Use the same Serial number to access GPS and MODEM. Please pay attention to thread safety.

MatiasRaya, 2025-04-25:
Sorry, I didn't understand the clarification.

lewisxhe, 2025-04-25:
Serial1 is an object. If you access it from multiple threads without a mutex, it can easily cause the program to crash. Search for FreeRTOS locks or check the examples provided by ESP.

MatiasRaya, 2025-04-25:
OK, thanks for your response.

lewisxhe, 2025-04-25:
Didn't you test it above? You have to place the antenna outdoors where you can see the sky.

closed 2025-05-13 by lewisxhe