Xinyuan-LilyGO/LilyGo-Modem-Series #262: GPS T-SIM7670G-S3

MatiasRaya, 2025-04-24:
Hello, I'm having trouble fixing the GPS antenna inside the store where it will be installed. How can I obtain the latitude and longitude coordinates?

MatiasRaya, 2025-04-24:
Attached log of the nmea output example:

Sats HDOP  Latitude   Longitude   Fix  Date       Time     Date Alt    Course Speed Card  Distance Course Card  Chars Sentences Checksum
           (deg)      (deg)       Age                      Age  (m)    --- from GPS ----  ---- to London  ----  RX    RX        Fail
----------------------------------------------------------------------------------------------------------------------------------------
AT+CGNSSPWR?
AT+CGNSSPWR?
+CGNSSPWR: 1**** ***** ********** *********** **** ********** ******** **** ****** ****** ***** ***   ******** ****** ***   8     0         0
0    0.0   ********** *********** **** 01/05/2080 23:59:42 411  ****** ****** ***** ***   ******** ****** ***   1338  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:43 442  ****** ****** ***** ***   ******** ****** ***   1718  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:44 444  ****** ****** ***** ***   ******** ****** ***   2098  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:45 445  ****** ****** ***** ***   ******** ****** ***   2478  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:46 445  ****** ****** ***** ***   ******** ****** ***   2858  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:47 446  ****** ****** ***** ***   ******** ****** ***   3238  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:48 446  ****** ****** ***** ***   ******** ****** ***   3618  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:49 447  ****** ****** ***** ***   ******** ****** ***   3998  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:50 451  ****** ****** ***** ***   ******** ****** ***   4378  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:51 452  ****** ****** ***** ***   ******** ****** ***   4758  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:52 452  ****** ****** ***** ***   ******** ****** ***   5138  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:53 453  ****** ****** ***** ***   ******** ****** ***   5518  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:54 454  ****** ****** ***** ***   ******** ****** ***   5898  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:55 454  ****** ****** ***** ***   ******** ****** ***   6278  0         0        
AT+CGNSSPWR?
AT+CGNSSPWR?
+CGNSSPWR: 10    0.0   ********** *********** **** 01/05/2080 23:59:56 462  ****** ****** ***** ***   ******** ****** ***   6666  0         0
0    0.0   ********** *********** **** 01/05/2080 23:59:57 464  ****** ****** ***** ***   ******** ****** ***   7046  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:58 464  ****** ****** ***** ***   ******** ****** ***   7426  0         0        
0    0.0   ********** *********** **** 01/05/2080 23:59:59 467  ****** ****** ***** ***   ******** ****** ***   7806  0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:00 469  ****** ****** ***** ***   ******** ****** ***   8186  0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:01 470  ****** ****** ***** ***   ******** ****** ***   8566  0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:02 471  ****** ****** ***** ***   ******** ****** ***   8946  0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:03 471  ****** ****** ***** ***   ******** ****** ***   9326  0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:04 470  ****** ****** ***** ***   ******** ****** ***   9706  0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:05 471  ****** ****** ***** ***   ******** ****** ***   10086 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:06 473  ****** ****** ***** ***   ******** ****** ***   10466 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:07 476  ****** ****** ***** ***   ******** ****** ***   10846 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:08 477  ****** ****** ***** ***   ******** ****** ***   11226 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:09 478  ****** ****** ***** ***   ******** ****** ***   11606 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:10 479  ****** ****** ***** ***   ******** ****** ***   11986 0         0        
AT+CGNSSPWR?
AT+CGNSSPWR?
+CGNSSPWR: 10    0.0   ********** *********** **** 01/06/2080 00:00:11 488  ****** ****** ***** ***   ******** ****** ***   12374 0         0
0    0.0   ********** *********** **** 01/06/2080 00:00:12 489  ****** ****** ***** ***   ******** ****** ***   12754 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:13 490  ****** ****** ***** ***   ******** ****** ***   13134 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:14 491  ****** ****** ***** ***   ******** ****** ***   13514 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:00:15 492  ****** ****** ***** ***   ******** ****** ***   13894 0         0        
[... cut ...]
0    0.0   ********** *********** **** 01/06/2080 00:01:51 636  ****** ****** ***** ***   ******** ****** ***   50422 0         0        
0    0.0   ********** *********** **** 01/06/2080 00:01:52 637  ****** ****** ***** ***   ******** ****** ***   50802 0         0        

It's a fragment. When I put the GPS antenna on the window, it took less than a minute to fix it. What improvements could I make to the antenna or how can I solve this problem, since the device is always in the middle of the room.

lewisxhe, 2025-04-25:
No GPS can locate indoors. All devices that can locate indoors rely on LBS.

lewisxhe, 2025-04-25:
Base station positioning, SIM7670G has no LSB

MatiasRaya, 2025-04-25:
Can AT command be sent to the GSM module to get the latitude and longitude of my cell phone operator?

lewisxhe, 2025-04-25:
Can't.

MatiasRaya, 2025-04-25:
And what method could I use to get my coordinates? I consult you since I need such data and in Argentina I only get the T-SIM7670G-S3

lewisxhe, 2025-04-25:
Either change to a model that supports LSB, or get the location via GNSS. There is no other way.

MatiasRaya, 2025-04-25:
How could I get a location via GNSS? Could it make it easier for me which AT commands I should use?

lewisxhe, 2025-04-25:
Didn't you test it above? You have to place the antenna outdoors where you can see the sky.

MatiasRaya, 2025-04-25:
Yes, I try it, but I have no possibility to remove the antenna since the.equipment it is mounted on is inside. Can the GPS antenna be replaced with one like the one in the image? 

[image]

lewisxhe, 2025-04-25:
Okay, I use this too.

MatiasRaya, 2025-04-25:
Do you know what the connector on the board to connect the GPS antenna is called? Or what is the .cable called to connect the sma to the board?

lewisxhe, 2025-04-25:
[image]
IPEX1 TO SMA

closed 2025-05-13 by lewisxhe