Xinyuan-LilyGO/LilyGo-Modem-Series #263: HTTP query using GPRS

MatiasRaya, 2025-04-25:
Hello, I am attaching the log of my application:

SD Card Type: SDHC
Tamaño: 14910MB
SD initialized successfully
[  3112][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=1 scl=2 freq=100000
[  3113][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[2025-04-25 20:12:48] [INFO] [main] [setup] RTC initialized successfully
[2025-04-25 20:12:48] [INFO] [main] [setup] RTC DateTime: 2025-04-25 20:12:48
[2025-04-25 20:12:48] [INFO] [led] [setLedState] LED Green OFF
[2025-04-25 20:12:48] [INFO] [led] [setLedState] LED Red OFF
[2025-04-25 20:12:48] [INFO] [led] [setLedState] LED Yellow OFF
[2025-04-25 20:12:48] [INFO] [led] [setLedState] LED Blue ON
[2025-04-25 20:12:48] [INFO] [main] [setup] SD initialized successfully
[2025-04-25 20:12:48] [INFO] [main] [setup] LittleFS mounted successfully
[  3229][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
[2025-04-25 20:12:48] [INFO] [config] [readConfig] Log rotation config: 5 files, 100 MB each
[2025-04-25 20:12:48] [INFO] [main] [setup] SN: 0001-00001
[2025-04-25 20:12:48] [INFO] [main] [setup] APN: igprs.claro.com.ar
[2025-04-25 20:12:48] [INFO] [main] [setup] SSID: Rayados, PSK: Rayaplasencia1996
[2025-04-25 20:12:48] [INFO] [main] [setup] Max files: 5, Max size: 100 MB
[2025-04-25 20:12:48] [INFO] [main] [setup] URL: thingsboard.iotmrpicomp.com, Port: 8080
[2025-04-25 20:12:48] [INFO] [main] [setup] Device ID: 4b2ed510-1a58-11f0-b5e1-f7b8eb07404e
[2025-04-25 20:12:48] [INFO] [main] [setup] Username: xx, Password: xxf
[2025-04-25 20:12:48] [INFO] [main] [setup] Config read successfully
[2025-04-25 20:12:48] [INFO] [thingsboard] [setData] Thingsboard URL and Device ID configured
[2025-04-25 20:12:48] [INFO] [main] [setup] Thingsboard data set successfully
[2025-04-25 20:12:48] [INFO] [main] [setup] APN configured, using custom APN
[2025-04-25 20:12:48] [INFO] [network] [initGSM] Start modem...
AT

QCRDY
AT
AT
OK

[2025-04-25 20:12:48] [INFO] [network] [initGSM] Waiting for network...
AT+CPIN?
AT
OK
AT+CPIN?
+CME ERROR: SIM busy

+CPIN: READY

SMS DONE
[2025-04-25 20:12:50] [INFO] [network] [initGSM] Network ready
AT+SIMCOMATI
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B02SIM767XM5A_M
SIM767XM5_B02V01_240717
IMEI: 864643060026905

OK
Set network apn : igprs.claro.com.ar
AT+CGDCONT=1,"IP","igprs.claro.com.ar"
AT+CGDCONT=1,"IP","igprs.claro.com.ar"
OK
[2025-04-25 20:12:50] [INFO] [network] [initGSM] Wait for the modem to register with the network.
.AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

[... 1263 lines cut ...]
    LOG_INFO(classNAME, "HTTP response code: %d", errorCode);

    if (errorCode == 200) {
        String response = httpClientPosition->responseBody();
        LOG_INFO(classNAME, "HTTP response body: %s", response.c_str());

        DynamicJsonDocument doc(1024);
        DeserializationError error = deserializeJson(doc, response);

        if (error) {
            LOG_ERROR(classNAME, "Failed to parse JSON: %s", error.c_str());
            return;
        }

        latitude = doc["loc"][0];
        longitude = doc["loc"][1];

        LOG_INFO(classNAME, "Latitude: %.6f, Longitude: %.6f", latitude, longitude);
    } else {
        LOG_ERROR(classNAME, "Failed to get coordinates from internet");
    }

    httpClientPosition->stop();
    httpClientPosition = nullptr;
}`

MatiasRaya, 2025-04-25:
What I don't understand is why the ping is successful, but when I try to make the HTTP request, it tells me it can't. I tested the SIM I'm using in a phone, and it works correctly.

MatiasRaya, 2025-04-26:
Using the httpclient example it gives me
[4211] Initializing modem...
[4211] ### TinyGSM Version: 1.0.0
[4211] ### TinyGSM Compiled Module: TinyGsmClientSIM7672
[4228] ### Modem: SIM7670G-MNGV
[4228] ### Modem: SIM7670G-MNGV
[4247] ### Modem: SIM7670G-MNGV
[4247] Modem Name: SIM7670G-MNGV
[4264] Modem Info: Manufacturer: SIMCOM INCORPORATED Model: SIM7670G-MNGV Revision: V1.9.05 IMEI: 864643060026905
Waiting for network...fail

And using the network example it gives me
Start modem...
AT
AT

QCRDY
AT
AT
OK

AT+CPIN?
AT
OK
AT+CPIN?
+CME ERROR: SIM busy

+CPIN: READY

SMS DONE
SIM card online
AT+SIMCOMATI
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B02SIM767XM5A_M
SIM767XM5_B02V01_240717
IMEI: 864643060026905

OK
Wait for the modem to register with the network.AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[6] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[7] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[8] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[9] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[10] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[11] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[12] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 99,99

OK
[13] Signal Quality:99
AT+CREG?
AT+CREG?
+CREG: 0,2

OK
AT+CSQ
AT+CSQ
+CSQ: 24,0

OK
[14] Signal Quality:24
AT+CREG?
AT+CREG?
+CREG: 0,6

OK
Registration Status:6

Registration Status:6
AT+CPSI?

+SMS FULL
AT+CPSI?
+CPSI: LTE,Online,722-310,0x5282,651276,107,EUTRAN-BAND7,3050,5,5,-9,-94,-84,17

OK
Inquiring UE system information:LTE,Online,722-310,0x5282,651276,107,EUTRAN-BAND7,3050,5,5,-9,-94,-84,17
AT+NETOPEN
AT+NETOPEN
OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR
+IPADDR: 10.13.133.202

OK
+IPADDR: 10.13.133.202
AT+CPING="www.google.com",1,1,64,1000,10000,255
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255
AT+CPING="www.google.com",1,1,64,1000,10000,255
ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255
AT+CPING="www.google.com",1,1,64,1000,10000,255
ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255
AT+CPING="www.google.com",1,1,64,1000,10000,255
ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255
AT+CPING="www.google.com",1,1,64,1000,10000,255
ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 2

+CPING: 3,1,0,1,0,0,0
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,199,119
Reply from ���?␏: bytes=64 time=199ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,199,199,199
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,30,119
Reply from ���?␏: bytes=64 time=30ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,30,30,30
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,42,119
Reply from ���?␏: bytes=64 time=42ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,42,42,42
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,47,119
Reply from ���?␏: bytes=64 time=47ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,47,47,47
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,43,119
Reply from ���?␏: bytes=64 time=43ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,43,43,43
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,42,119
Reply from ���?␏: bytes=64 time=42ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,42,42,42
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,42,119
Reply from ���?␏: bytes=64 time=42ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,42,42,42
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119
AT+CPING="www.google.com",1,1,64,1000,10000,255

+CPING: 3,1,1,0,45,45,45
AT+CPING="www.google.com",1,1,64,1000,10000,255
OK

+CPING: 1,142.250.79.100,64,45,119
Reply from ���?␏: bytes=64 time=45ms TTL=119

+CPING: 3,1,1,0,45,45,45

lewisxhe, 2025-04-26:
Your sketch is too complex, please use the minimum example test https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsBuiltlnGet

lewisxhe, 2025-04-26:
Please keep your question short and simple instead of copying it all to give me a headache.

lewisxhe, 2025-04-26:
It is meaningless to use multithreading in TinyGSM, because all data is accessed by a Serial, and your threads are not safe, and data is even sent out of order. Please learn how to safely use FreeRTOS threads to safely access the same object.

MatiasRaya, 2025-04-26:
The threads are not being used, notice that the statements of the TASK in the STATMENT are commented.a.

MatiasRaya, 2025-04-28:
Hi, I'm still having trouble using the SIM card on my board. I've attached the log of the execution. One thing I don't understand is why it authenticates with a 6 (roaming network) instead of using the local data network.

SD Card Type: SDHC
Tamaño: 14910MB
SD initialized successfully
[  3055][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=1 scl=2 freq=100000
[  3056][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[2025-04-28 19:01:03] [INFO] [main] [setup] RTC initialized successfully
[2025-04-28 19:01:03] [INFO] [main] [setup] RTC DateTime: 2025-04-28 19:01:03
[2025-04-28 19:01:04] [INFO] [led] [setLedState] LED Green OFF
[2025-04-28 19:01:04] [INFO] [led] [setLedState] LED Red OFF
[2025-04-28 19:01:04] [INFO] [led] [setLedState] LED Yellow OFF
[2025-04-28 19:01:04] [INFO] [led] [setLedState] LED Blue ON
[2025-04-28 19:01:04] [INFO] [main] [setup] SD initialized successfully
[2025-04-28 19:01:04] [INFO] [main] [setup] LittleFS mounted successfully
[  3182][W][LittleFS.cpp:77] begin(): LittleFS Already Mounted!
[2025-04-28 19:01:04] [INFO] [config] [readConfig] Log rotation config: 5 files, 100 MB each
[2025-04-28 19:01:04] [INFO] [main] [setup] SN: 0001-00001
[2025-04-28 19:01:04] [INFO] [main] [setup] APN: igprs.claro.com.ar
[2025-04-28 19:01:04] [INFO] [main] [setup] Max files: 5, Max size: 100 MB
[2025-04-28 19:01:04] [INFO] [main] [setup] Config read successfully
[2025-04-28 19:01:04] [INFO] [network] [setData] Thingsboard URL and Device ID configured
[2025-04-28 19:01:04] [INFO] [main] [setup] Thingsboard data set successfully
[2025-04-28 19:01:05] [INFO] [main] [setup] APN configured, using custom APN
[2025-04-28 19:01:05] [INFO] [network] [initGSM] Start modem...
AT

QCRDY

+CPIN: READY

SMS DONE
AT
OK

AT
AT
OK
ATE0
ATE0
OK
AT+CMEE=0

OK
AT+CTZR=0

OK
AT+CTZU=1

OK
AT+CPIN?

+CPIN: READY

OK
[2025-04-28 19:01:05] [INFO] [network] [initGSM] Waiting for network...
AT+CPIN?

+CPIN: READY

OK
[2025-04-28 19:01:05] [INFO] [network] [initGSM] Network ready
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B05SIM767XM5A_M
SIM767XM5_B05V01_241206
IMEI: 864643060026905

OK
Set network apn : igprs.claro.com.ar
AT+CGDCONT=1,"IP","igprs.claro.com.ar"

OK
[2025-04-28 19:01:05] [INFO] [network] [initGSM] Wait for the modem to register with the network.
.AT+CREG?

+CREG: 0,2

OK
AT+CSQ

+CSQ: 27,0

OK
[2025-04-28 19:01:05] [INFO] [network] [initGSM] Signal Quality: 27
.AT+CREG?

+CREG: 0,6

OK
Registration Status: 6

[2025-04-28 19:01:07] [INFO] [network] [initGSM] Registration Status: 6
AT+CPSI?

+SMS FULL

+CPSI: LTE,Online,722-310,0x5282,651276,107,EUTRAN-BAND7,3050,5,5,-10,-89,-79,19

OK
[2025-04-28 19:01:08] [INFO] [network] [initGSM] Inquiring UE system information: LTE,Online,722-310,0x5282,651276,107,EUTRAN-BAND7,3050,5,5,-10,-89,-79,19
AT+NETOPEN

OK

+NETOPEN: 0[2025-04-28 19:01:08] [INFO] [network] [initGSM] Network active
AT+IPADDR

+IPADDR: 10.77.113.30

OK
[2025-04-28 19:01:13] [INFO] [network] [initGSM] Network IP: +IPADDR: 10.77.113.30
AT+CSQ

+CSQ: 27,0

OK
[2025-04-28 19:01:13] [INFO] [network] [initGSM] Network Signal Quality: 27
AT+NETOPEN?

+NETOPEN: 1

OK
AT+IPADDR

+IPADDR: 10.77.113.30

OK
[2025-04-28 19:01:13] [INFO] [network] [initGSM] GPRS is connected
AT+CPING="www.google.com",1,1,64,1000,10000,255

OK
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
AT+CPING="www.google.com",1,1,64,1000,10000,255

ERROR
Error code : -1
[2025-04-28 19:01:23] [ERROR] [network] [initGSM] Ping failed!

lewisxhe, 2025-04-29:
Have you tested changing the sim card?

MatiasRaya, 2025-04-29:
I tried another sim and had the same result. It could be due to the error that CPING misarms the package to send it to google? I consult because it changes www.google.com for 8.8.8.8 and if I pinged, and when making a query to a website that returns information about gps location I do well with the information, so it seems to be google. The issue is that I tried with the AT commands by hand and I was able to ping Google that way, the AT command I used was
AT+CPING="www.google.com",1,4,64,1000,10000,255

lewisxhe, 2025-04-30:
It seems that the parameter of num_pings is wrong. The default num_pings is set to 4. If you can ping with the default value, then the network is normal.

MatiasRaya, 2025-04-30:
perfect, I do the test and warn you. still, that value is in the code of the bookstore, but I make the change and inform you

MatiasRaya, 2025-05-01:
hello, I tested with the command AT+CPING="www.google.com,"1,4,64,1000,10000,255 and I got wrong, put a 1 back and walked around

closed 2025-05-02 by MatiasRaya