Xinyuan-LilyGO/LilyGo-Modem-Series #264: Can't read the same sms on diferents boards A7670G

alexlopez2612, 2025-04-27:
Hello
Like 6 months ago I got a board with A7670G and I developed a code that can read sms.
This is an example for a sms that I can read correctly on my old board:

_AT+CMGRD=1
+CMGRD: "REC UNREAD","+34626704066","","25/04/27,13:44:52+8"
1234 estado_ 

The info of that old A7670G is:
_AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A011B19A7670M7
A7670M7_B19V01_240126
QCN:
IMEI: 867284063580528
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:_

But, I just got another board, and on this  board, the sms, with the same sim and the same sms from the same phone, i get this:

_AT+CMGRD=1
+CMGRD: 0,"",30
07914346220050F0040B914326764060F60000524072319385800C31D98C062ACFE961F21B04_

The info of this new board is:
_Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A011B20A7670M7
A7670M7_B20V01_240325
QCN:
IMEI: 867284065516041
MEID:
+GCAP: +CGSM,+FCLASS,+DS
DeviceInfo:_

As you can see, the info that I get reading de sms is incorrect, I can't see the number, the date, and the content of the sms.

In the old board I never saw this... What it's going on?

Thank you on advance!

Regards

lewisxhe, 2025-04-28:
Are the output formats of the `AT+CMGF?` query text messages sent by two boards the same?

alexlopez2612, 2025-04-28:
Hello @lewisxhe 
Thank you very much for your reply. That was the problem. This part of the code I wrote it like 6 months ago, and now, looking for the place where to put the command that you said, I saw that command commented. 
I think that was uncommented at first and the AT+CMGF=1 was saved in the flash of A7670G, and that's the reason that keep working, rigth?
Now I excecute AT+CMGF=1 on the new one and works perfectly. Thank you very much.
A question. I is bad for A7670G that launch this command every time that I turn on the circuit? I'm asking that because also the apn AT command Im launching it every time and I don't know if this is bad for the memory of the chip.

Thank you and regards

lewisxhe, 2025-04-29:
Starting it each time does not affect anything.

alexlopez2612, 2025-04-29:
Hello @lewisxhe 
Thank you!!!!

closed 2025-04-29 by alexlopez2612