Xinyuan-LilyGO/LilyGo-Modem-Series #265: Esp32 keeps awakening by MODEM_RING_PIN

JoanSalelles, 2025-04-28:
Hi!
I'm trying to develop an IoT application in which I need to put the ESP32 in deep sleep and then waking it up either by a timer or when a SMS is received.
In my preliminar tests the timed awakening is working well but I can manage the awakening by GPIO work.
I'm using the MODEM_RING_PIN that in my Lyligo T-A7670E is GPIO33 as the waking pin in order to wake up the ESP32 by it going low when an SMS is received, but it doesn't work, the ESP32 wakes up constantly just after being put in deep sleep, withount having any SMS been sent.
My code is:
`/**
 * @file      ReadSMS.ino
 * @author    Lewis He (lewishe@outlook.com)
 * @license   MIT
 * @copyright Copyright (c) 2024  Shenzhen Xin Yuan Electronic Technology Co., Ltd
 * @date      2024-08-29
 * @note      SIM7670G does not support SMS and voice functions
 * ! Only read data, not decode data. For detailed SMS operation, please refer to A76XX_Series_AT_Command_Manual
 */

#include "utilities.h"
//Per manipular els RTC_GPIO
#include "driver/rtc_io.h"

#define SMS_TARGET  "+34685817535"

#define uS_TO_S_FACTOR 1000000ULL  // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  25           // Time ESP32 will go to sleep (in seconds)

RTC_DATA_ATTR int bootCount = 0;

// Method to print the reason by which ESP32 has been awaken from sleep
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

String data;
int8_t res;

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

#include <TinyGsmClient.h>

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

void configuraModem() {
    // Turn on DC boost to power on the modem
    #ifdef BOARD_POWERON_PIN
        pinMode(BOARD_POWERON_PIN, OUTPUT);
        digitalWrite(BOARD_POWERON_PIN, HIGH);
[... 299 lines cut ...]
AT+CMGL="ALL"
Read all message failedets Jul 29 2019 12:21:46

rst:0x5 (DEEPSLEEP_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4888
load:0x40078000,len:16516
load:0x40080400,len:4
load:0x40080404,len:3476
entry 0x400805b4
Wakeup caused by external signal using RTC_IO
Boot number: 3
Despertat per RI!
AT+CMGL="ALL"
Read all message failedets Jul 29 2019 12:21:46
`
And it keeps awakening right after being put in deep sleep.

I've even tried to put an externalphysical pull up resistor of 10K to the pin 33 of the board.

Please I beg you to help me, because I've been stuck with this for weeks!

Thanks in advance.

lewisxhe, 2025-04-29:
I didn't see you keep the modem turned on before sleep, so when esp32 is in sleep, the modem is also powered off and can't wake up. Please keep GPIO12 before sleep.

Just like this example does
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/d8ce72586b1a26453e72abda6c7fe5e62a6eedfe/examples/ModemSleep/ModemSleep.ino#L129

JoanSalelles, 2025-04-29:
> I didn't see you keep the modem turned on before sleep, so when esp32 is in sleep, the modem is also powered off and can't wake up. Please keep GPIO12 before sleep.
> 
> Just like this example does
> 
> [LilyGO-T-A76XX/examples/ModemSleep/ModemSleep.ino](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/d8ce72586b1a26453e72abda6c7fe5e62a6eedfe/examples/ModemSleep/ModemSleep.ino#L129)
> 
But my problem is just the opposite: the ESP32 wakes up even though the modem hasn't received any SMS.

When the ESP32 goes to deep sleep, it is immediately awaken and the reason of the awakening reported by _esp_sleep_get_wakeup_cause()_ is ESP_SLEEP_WAKEUP_EXT0, that is the MODEM_RING_PIN (GPIO33) going LOW even thoug I have it pulled up with an external 10K resistor.

By the way, I also tried to put the modem to sleep with the example ModemSleep.ino you mentioned and it doesn't work either, but this will be a second stage on my project developing. Now I only would like to put the ESP32 in deep sleep and the modem to wake it up when receiving an SMS.

lewisxhe, 2025-04-29:
You misunderstood. What I meant is that before you set the ESP to deep sleep, you should keep the power control IO at a high level. Instead of going directly to sleep, otherwise the modem will lose power.

digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
gpio_hold_en((gpio_num_t)MODEM_RESET_PIN);
gpio_deep_sleep_hold_en();
gpio_hold_en((gpio_num_t )BOARD_POWERON_PIN);
gpio_deep_sleep_hold_en();

Put these lines before sleep and execute

lewisxhe, 2025-04-29:
I added example, please check https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/WakeupByRingOrSMS

JoanSalelles, 2025-04-30:
> I added example, please check https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/WakeupByRingOrSMS

It works!!!!!

The code you provided worked right at the first time! Now I'm going to adapt your example code to my project!

Than you, thank you very much, Mr. @lewisxhe !!! I have been stuck with this for days!

closed 2025-04-30 by JoanSalelles