Xinyuan-LilyGO/LilyGo-Modem-Series #268: Device with LBS

MatiasRaya, 2025-04-30:
Hello, I wanted to know if the image board has LBS to make the purchase of it 

[image]

MatiasRaya, 2025-05-02:
Perfect, thanks for your recommendation.

MatiasRaya, 2025-05-02:
@lewisxhe would this be the device you tell me? 

https://lilygo.cc/products/t-sim-a7670e?variant=42750698422453

De ser así, sabe si hacen envío a Argentina?

lewisxhe, 2025-05-02:
If you are in South America, you should choose a7670sa. And the version with GPS. As for when it can be delivered, you need to ask the salesperson.

MatiasRaya, 2025-05-02:
What would be the South American version? will be the one attached in this image? 

[image]

lewisxhe, 2025-05-05:
Yes !

closed 2025-05-13 by lewisxhe