Xinyuan-LilyGO/LilyGo-Modem-Series #270: Cant detect my modem until now

Makuu6, 2025-05-01:
[image]
[image]

#include "utilities.h"
#include "Arduino.h"
#include <TinyGsmClient.h>
#include <TinyGPSPlus.h>

#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1

TinyGsm modem(SerialAT);
TinyGPSPlus gps;

uint32_t check_interval = 0;

String message = "this is Live Tracking Coordinates from Project ARIA";  // Global message

bool isSendingSMS = false; // Flag to avoid interference between GPS data and SMS sending

void setup() {
    Serial.begin(115200); // Set console baud rate
    Serial.println("🚀 Start Sketch");

    SerialAT.begin(115200, SERIAL_8N1, MODEM_RX_PIN, MODEM_TX_PIN);

    // Reset modem
    pinMode(MODEM_RESET_PIN, OUTPUT);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);
    delay(100);
    digitalWrite(MODEM_RESET_PIN, MODEM_RESET_LEVEL);
    delay(2600);
    digitalWrite(MODEM_RESET_PIN, !MODEM_RESET_LEVEL);

    // Power ON the modem
    pinMode(BOARD_PWRKEY_PIN, OUTPUT);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, HIGH);
    delay(100);
    digitalWrite(BOARD_PWRKEY_PIN, LOW);

    if (AutoBaud()) {
        Serial.println(F("***********************************************************"));
        Serial.println(F(" 📡 Modem Ready! "));
    } else {
        Serial.println(F("***********************************************************"));
        Serial.println(F(" ❌ Failed to connect to the modem! Check the baud rate."));
        Serial.println(F("***********************************************************\n"));
        return;
    }

    // Initialize GPS
    Serial.println("Start modem...");
    delay(3000);

    Serial.print("Modem starting...");
    while (!modem.testAT(1000)) {
        Serial.print(".");
    }
    Serial.println();

    Serial.println("Enabling GPS...");
    while (!modem.enableGPS(MODEM_GPS_ENABLE_GPIO, MODEM_GPS_ENABLE_LEVEL)) {
        Serial.print(".");
    }
    Serial.println("GPS Enabled");

    modem.setGPSBaud(115200);
    modem.setGPSMode(7);  // Set GPS mode to use all available satellites (GPS, GLONASS, etc.)
    modem.enableNMEA();

    // ✅ Update SMSC smart
    SerialAT.println("AT+CSCA=\"+639180000101\"");
    delay(1000);

    // List of recipient phone numbers
    String users[] = { "+639949647812","+639627003611", "+639098206044" };

    // SMS Message
    String broadcastMessage = "This is a test broadcast message! YOU HAVE BEEN WARNED!";

    // Send messages to all recipients
    for (int i = 0; i < sizeof(users) / sizeof(users[0]); i++) {
        sendSMS(users[i], broadcastMessage);
    }
}

void loop() {
    // Handle Serial Input
    if (Serial.available()) {
        String input = Serial.readString();
        input.trim();  // Remove any extra spaces or newlines

        if (input == "SEND GPS" && !isSendingSMS) {
            String gpsLocation = getGPSLocation();  // Get GPS location as a string

            // List of recipient phone numbers
            String users[] = {"+639949647812", "+639627003611", "+639098206044"};

            // Send GPS location to all numbers in the array
            for (int i = 0; i < sizeof(users) / sizeof(users[0]); i++) {
                sendSMS(users[i], gpsLocation + " " + message);  // Send GPS location along with the message
            }
        }
    }

    smartDelay(2000);
}

// Function to format GPS location into a readable string
String getGPSLocation() {
    if (gps.location.isValid()) {
        String lat = String(gps.location.lat(), 6);
        String lng = String(gps.location.lng(), 6);
        String location = "Latitude: " + lat + ", Longitude: " + lng;
        return location;
    } else {
        return "Invalid GPS Location!";
    }
}

void sendSMS(String number, String message) {
    Serial.print("📨 Sending message to ");
    Serial.println(number);

    isSendingSMS = true; // Flag to prevent GPS data overlap

    // Send SMS command
    SerialAT.print("AT+CMGS=\"");
    SerialAT.print(number);
    SerialAT.println("\"");
    delay(500); // Wait for prompt

    SerialAT.print(message);  // Send message content
    delay(200);

    SerialAT.write(0x1A);  // Send CTRL+Z to submit SMS
    delay(1000);  // Wait for response

    // Optional: you can comment this out if no response logging is required
    // String response = SerialAT.readString();
    // Serial.println("📡 Modem Response: " + response);

    // Simple success message
    Serial.println("✅ Message sent successfully!");

    isSendingSMS = false; // Reset flag after sending SMS
}

uint32_t AutoBaud() {
    SerialAT.updateBaudRate(115200);
    delay(100);

    for (int j = 0; j < 10; j++) {
        SerialAT.print("AT\r\n");
        String input = SerialAT.readString();
        if (input.indexOf("OK") >= 0) {
            Serial.println("✅ Modem is responding at 115200 baud.");
            return 115200;
        }
    }

    Serial.println("❌ Failed to communicate with the modem at 115200 baud.");
    return 0;
}

static void smartDelay(unsigned long ms) {
    unsigned long start = millis();
    while (millis() - start < ms) {
        while (SerialAT.available()) {
            gps.encode(SerialAT.read());
        }
    }
}
i cant detect my module
7670G on device manager 
just had conflict with arduino attached whil uploading so it happens that it didnt accept compiling due to conflicting Arduino.h
now i cant detect my module

lewisxhe, 2025-05-05:
Which pins did you use? Have you tried disconnecting all externally connected GPIOs?

Makuu6, 2025-05-05:
yes ive tried ...no connections .. the serial wont display in my arduino
IDE and it keeps on disconnecting USB Serial

lewisxhe, 2025-05-06:
Continuously disconnecting USB serial?

This is a problem that occurs when the program is running abnormally. ESP32S3 does not have the problem of not being able to upload sketches.

Have you entered download mode?
[image]

Makuu6, 2025-05-06:
i tried combo with RST and Ground

lewisxhe, 2025-05-06:
I see your picture is COM1? Are you sure this device is COM1?

Open the device manager and then insert the USB. Will there be any changes on the COM side?

Makuu6, 2025-05-06:
the Usb serial automatically disconnects after upload and when plug back
the usb it wont run normally.

lewisxhe, 2025-05-06:
Can you upload a simple program to print serial to the ESP and test it? It would be best if there is a video so I can understand what you did.

Makuu6, 2025-05-06:
okay ill send a video later..thanks

Makuu6, 2025-05-06:
How to send a video here ...i had 4 mins of video its big file.

Makuu6, 2025-05-06:
I tried simple code like initial GPS it wont display anything from Serial
monitor

Makuu6, 2025-05-06:
[image]

lewisxhe, 2025-05-07:
> How to send a video here ...i had 4 mins of video its big file.
> 怎么在这里发送视频……我有 4 分钟的视频，文件很大。
> […](#)

Upload the video to youtube.

closed 2025-06-26 by github-actions[bot]