Xinyuan-LilyGO/LilyGo-Modem-Series #271: [Bug]: A7670G Modem not working after update

edwinbrowwn, 2025-05-02:
### Category

Other

### Hardware

T-A7670

### PN Number

S2-10AZT-232GV

### Firmware Version

```Shell
A124B01A7670M7
```

### Description

OLD OUTPUT:
```
AT+SIMCOMATI
Manufacturer: INCORPORATED
Model: A7670G-LLSE
Revision: A011B20A7670M7
A7670M7_B20V01_240325
QCN:
IMEI: 867284065530893
```

NEW OUTPUT:
```
AT+SIMCOMATI
Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 
```

### Relevant log output

```Shell
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Start Sketch
Trying baud rate 115200
Trying baud rate 9600
Trying baud rate 57600
Trying baud rate 38400
Trying baud rate 19200
Trying baud rate 74400
Trying baud rate 74880
Trying baud rate 230400
Trying baud rate 460800
Trying baud rate 2400
Trying baud rate 4800
Trying baud rate 14400
Trying baud rate 28800
***********************************************************
 Failed to connect to the modem! Check the baud and try again.
***********************************************************
```

lewisxhe, 2025-05-02:
Please try to update the firmware again

edwinbrowwn, 2025-05-03:
I'm still getting the same exact issue. I have installed 3 times so far @lewisxhe

lewisxhe, 2025-05-05:
Use micro-usb to connect the USB port of A7670 modem, select USB-AT port, and then send AT command to test whether it can return data!

edwinbrowwn, 2025-05-09:
[image]
After re-flashing several times, I still have no network access on the modem.

lewisxhe, 2025-05-09:
I just found out today that A76XX_A79XX_MADL V1.13 has a bug that causes the IMEI to be lost. Maybe this is the problem of the network not being able to be turned on. Please use the latest tool to re-flash the firmware.

After flashing, please use AT+SIMEI= your IMEI number on the modem label

Then re-test it

https://drive.google.com/file/d/12nt5-wcsUT6bRaEhfOMBSq0EhOl8R2by/view?usp=sharing